tuya-cloudcutter/tuya-cloudcutter #344: XR809 support

Wolfieeewolf, 2023-03-30:
I have a few door sensors that have XR3 chips in them. I have been struggling to flash these manually. The few documented flashing guides I have found are a little confusing to me. I am pretty sure I'm doing it right but just can't seem to get it to flash. 

What are the possibilities for adding XR809 chip support to the cloud cutter app? I will keep trying to flash the manual way for now but it would be a whole lot easier to do with cloud cutter.

Wolfieeewolf, 2023-04-01:
The XR Chips are ARM based - https://developer.tuya.com/en/docs/iot/xr-series-modules?id=Kaiuyiuke8fu1

I know for sure that the XR3 chip is ARM based - https://developer.tuya.com/en/docs/iot/xr3-datasheet?id=K98s9168qi49g

Unfortunately, I have been unsuccessful if dumping the base firmware. The method of flashing the device requires something like 7 flashing pins to be connected at once and the use of a program called PhoenixMC to flash the device. which is not in English so it's a matter of guessing. 

There have been some successful flashing with devices over on the elektroda forums.

https://www.elektroda.com/rtvforum/topic3806769.html
https://www.elektroda.pl/rtvforum/topic3771510.html
https://www.elektroda.pl/rtvforum/topic3890640.html

Some of these links are not in English but using TWP - Translate Web Pages extension for firefox help fix the problem

If I do manage to get the firmware for you I will upload it here. For now, I'm stuck.

johnpwhite, 2024-01-14:
Hi, just checking to see if this moved forward etc.?

Cossid, 2024-01-14:
No progress, no dumps have been provided to be analyzed.  I have not yet been able to procure a Tuya device with this chip yet either.

johnpwhite, 2024-01-16:
Where are you based? I'm in Australia and could maybe ship mine?

I'm ok with electronics and might be able to create a dump, if you can suggest low cost tech and a guide?

Cossid, 2024-01-16:
I'm based in the US, but there are many things we need to know before direct hardware even becomes useful, getting a dump is the first step.  There are many things we need to work out first, like encryption (it appears XR809 is unencrypted, but I'm not certain so far), Tuya SDKs and debug files, protocols present in SDKs, OTA formats, and ARM instructions available.  There is no guarantee this is even possible.

I have no experience with XR809, it looks like https://www.elektroda.com/rtvforum/topic3806769.html would be the best resource for getting the tools to potentially create a dump.

divadiow, 2024-09-17:
https://www.elektroda.com/rtvforum/viewtopic.php?p=21146063#21146063