tuya-cloudcutter/tuya-cloudcutter #346: Changing WiFi SSID After cutting

zoidy, 2023-04-01:
I've successfully cut a [Lenovo bulb](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/lenovo-se-242dc-rgbct-bulb-v1.2.21.json) but have not flashed any custom firmware. My understanding is that if I need to change the Wi-Fi that the bulb connects to, I need to rerun the cutting process, specifying a new SSID and password.

During the initial cut, the bulb did not automatically connect to the cloudcutterflash AP and I had to use the Tuya app to manually tell the bulb to connect to it. Now that it's cut, I obviously can't use the official app anymore. What is the process to manually tell the device to connect to cloudcutterflash in case I re-run the cutting process and it doesn't autoconnect? Is there another way to change the information so that if I change my home network's password or the bulb gets accidentally reset, I don't end up with a paperweight?

Cossid, 2023-04-01:
You actually can use the app for that part still.  All that is happening is Smart Life connects directly to the bulb then sends data for how to connect to a network.  It's everything after that part that would not work anymore.

zoidy, 2023-04-02:
Ah great. I assume that the name of the AP is still SmartLife-xxxx, even after cutting? That's what the app is telling me it expects.

Cossid, 2023-04-02:
No, the data you need to send is for CloudCutter which is ssid `cloudcutterflash` and password `abcdabcd` and the AP you need to connect to once already cut with be `A-XXXX` instead of `SmartLife-XXXX`  After you connect to the A-, there should be something like a 'confirm hotspot' option if it doesn't like the name.

zoidy, 2023-04-02:
Thanks for the quick replies. I understand I need to send the `cloudcutterflash` info to the device. That's what I did the first time - though I suppose I didn't use actually AP mode to send the info because I don't remember connecting my phone to anything special.  The cut was successful because I did get the keys (and they work in LocalTuya in Home Assistant)

But you did answer my question about the name of the AP to connect to. I think I understand now. From what you describe, the Tuya app doesn't care that the AP of the bulb is named `A-xxx` instead of `SmartLife-xxx` when connecting. Therefore, it will always be possible for the app to connect to the cut device and send the `cloudcutterflash` info (assuming the app itself isn't patched to detect cut devices)

Cossid, 2023-04-02:
Correct, it might give you a screen to go back to device connection, but it should also have an option just below that to confirm the hotspot you are connected to (sometimes it just continues on without that though).

You can also use the 3x quick-blink mode and use Smart Life you send manually that way as well (so you don't have to connect to the AP at all).

closed 2023-04-02 by zoidy

zoidy, 2023-04-04:
Issue #316 has been incredibly helpful with understanding my question . I had similar problems with it not properly connecting to the `cloudcutterflash` wifi.  Based on @Cossid 's [hint ](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/316#issuecomment-1465307244) about how cloudcutter changes the wifi credentials. I was able to take that function and do some light modifications so it runs standalone. 

> I have tried extending the tinytuya timeout values a bit to see if that helps any. If you ever get a 900 or 904 error from tinytuya, your device is stuck in activation mode.

I had to increase tinytuya's timeout or else I would get the error `{'Error': 'Timeout Waiting for Device', 'Err': '902', 'Payload': 'Check device key or version'}` (I picked 3.2 instead of 0.2 for the timeout). 

I was then able to change the wifi credentials without going through the whole cutting process. There are a few bugs to work out as it didn't connect right away (router rejected the PSK, even though it was correct. I had to temporarily create an unencrypted AP and restart the light a few times).