tuya-cloudcutter/tuya-cloudcutter #349: New device

tilator, 2023-04-04:
Hi,

I just want to inform you, that Nedis WIFIP121FWT can be flashed using QNCX 16A plug configuration.

WBR:  Tilator

Cossid, 2023-04-04:
Clarified here that Nedis WIFIP121FWT is compatible with QNCX 16A Smart Plug profile - 1.1.8 / oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00

We don't add a new device to the list unless we verify the entire profile.  This information is valid for 3rd party firmware flashing only.  Closing as it is documented, there is no issue to address.

closed 2023-04-04 by Cossid

ile6695, 2025-04-17:
Here's the original firmware for the device: [https://archive.org/details/wifip-121-fwt](https://archive.org/details/wifip-121-fwt)

Cossid, 2025-04-17:
Added as `Nedis WIFIP121FWT Plug v1.1.8`