tuya-cloudcutter/tuya-cloudcutter #350: Feit 6" Smart Wi-Fi LED Ultra-Thin Canless Downlight - BK7231N - Firmware dump attached (OpenBeken Flasher read from UART)

synssins, 2023-04-07:
[Feit_Canless_6in_RGBCW.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11175659/Feit_Canless_6in_RGBCW.zip)

Brand: Feit Electric
Model Number: LEDR6XT_RGBW_CA_AG
Description: Feit Electric 6" Smart Wi-Fi LED Ultra-Thin Canless Downlight 
Type: RGBCW, direct PWM

Module: BK7231N
Module type: CBU (marked on board)
PWM Channels:
R = 26
G = 24
B = 6
C = 8
W = 7

[image]

Good evening. I am not familiar with submitting GitHub issues, so please let me know if I am missing anything in this submission. I'm adding everything I feel is relevant. 

I have a dozen of these lights that I'd like to flash OpenBeken to, and will be submitting documentation on the teardown of the first unit I performed. The teardown requires cutting rivets, so I would love to be able to flash OTA with Tuya-Cloudcutter. 

I dumped the firmware from the first unit using the OpenBeken flash app via UART, then dumped it a second time at a second bitrate and hashed the two files. They hashed the same, so I know I got a clean dump. I was having some issues with my RX/TX pins... My eyes aren't what they used to be. and the pins are tiny. 
Please see the attached firmware zip.

OpenBeken flashed cleanly and the light came up immediately.

What else is needed to build a profile? I can crack a second, unmodified light and hit the UART again.
Additionally, if there's a submission template I missed on what information is needed, please point me at it.

Cossid, 2023-04-07:
New device added as `Feit LEDR6XT-RGBW-CA-AG 6-Inch Canless Downlight`

closed 2023-04-07 by Cossid

synssins, 2023-04-07:
@Cossid that was an incredibly prompt response. Wow!

Sadly, that did not seem to work for me. 
What am I missing?

[image]

synssins, 2023-04-07:
Disregard. I rebooted everything and it's good now.

Cossid, 2023-04-07:
Also make sure to update your git repo, there was a bug a couple days ago that caused you have to have a really quick device reset or it would prematurely fail.