tuya-cloudcutter/tuya-cloudcutter #355: Update devices not in the list

fokcuk, 2023-04-13:
Sorry, no Q&A section that I could find or a forum of some kind
I have Arlec Grid Connect 4 way power point. Its using BK7231T WB2S chipset. Just has 4 relays and an on/off button.

Do I understand correctly, that I should be able to use any profile that is made for WB2S to upload custom firmware? I

ts already documented on OpenBK7231T with all pins docunted

Cossid, 2023-04-13:
No, that is not correct.  Module is far too generic, you need a matched profile, which will relate to a specific firmware version and device class usually.  See https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has to find your firmware version, then use `by firmware version and name`

fokcuk, 2023-04-14:
thank you! I checked the FAQ and it pointed me in a direction of checking MAC addresses - turns out that I have an older version that still runs on ESP chip, so was able to use Tuya-convert.

closed 2023-04-14 by fokcuk