tuya-cloudcutter/tuya-cloudcutter #357: New device: LEDVance wifi remote control

stofakiller, 2023-04-14:
I just (very easy) took it apart and it has a BK7231N chip. I think it could be nice to use it through mqtt to eg Home Assistant and others…

[image]
[image]
[image]
[image]

Cossid, 2023-04-14:
We can't do anything without a firmware dump.  Remotes have highly specific low-power firmware, and cutting them probably isn't going to be terribly useful.  There is nothing actionable here, so closing this issue.

closed 2023-04-14 by Cossid

stofakiller, 2023-04-17:
Hi @Cossid,

I know this one is closed, but i can't see how to else send a comment to you

I now have flashed OpenBK7231N on it, and have the backup bin file of the original firmware... Would it be possible have another look at it?

Cossid, 2023-04-17:
You're welcome to post the original firmware, and I will add a profile if it is unpatched, but even though a profile is possible to make, the way these devices are powered means CloudCutter won't really work on it.