tuya-cloudcutter/tuya-cloudcutter #361: [CB3S] [BK7231N] Eacam 24W RGB Ceiling Lamp

ataa, 2023-04-20:
PCB: SK-M24WLED-TY/C/E-V3

[image]
[Ceiling-Down-light-sk-m24wled-ty-c-e-v3-readResult_BK7231N_QIO_2023-19-4--14-49-16.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11284922/Ceiling-Down-light-sk-m24wled-ty-c-e-v3-readResult_BK7231N_QIO_2023-19-4--14-49-16.zip)

Cossid, 2023-04-20:
Added as `Tuya Generic SK-M24WLED-TY-C-E-V3 24W RGBCT Ceiling Light`.  Unfortunately, this device doesn't have configuration data in storage.

closed 2023-04-20 by Cossid

ataa, 2023-04-20:
Found the pins by checking the traces:

```json
  "pins": {
    "6": "PWM;4",
    "7": "PWM;3",
    "8": "PWM;0",
    "24": "PWM;1",
    "26": "PWM;2"
  },
```

banter7702, 2023-10-27:
Hi, sorry for opening this.
I have the same ceiling lamp with the same pcb board.

Yours must have remote control as well?

I managed to flash esphome using cloudcutter. However, I'm contemplating if I made good decision.

Can't figure out how to restore remote functionality, or make functionalities as stock as possible.

Would you mind sharing your yaml/config?