tuya-cloudcutter/tuya-cloudcutter #362: Failed to build Docker image

Hotelk52339, 2023-04-20:
Hello

I have a problem running the tuya-cloudcutter, i'm using Kali Linux 

`(kali㉿kali)-[~/tuya-cloudcutter]
└─$ sudo ./tuya-cloudcutter.sh                               

Building cloudcutter docker image
common.sh: line 113: docker: command not found
Failed to build Docker image, stopping script`

Anyone have solution for that ?

kuba2k2, 2023-04-20:
You don't have Docker installed, see [INSTRUCTIONS.md](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#prerequisites).

Hotelk52339, 2023-04-20:
Yep, It worked! 
Thank you

closed 2023-04-20 by Hotelk52339

amedeodivito, 2023-10-29:
I can't complete the operations
I use Raspberry pi3
I follow all the instructions but when I give the command
sudo ./tuya-cloudcutter.sk
after a while it ends with errors

`pi@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -r
Building cloudcutter docker image
[+] Building 70.2s (6/12)                                                                                                                                                                               docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                                              0.1s
 => => transferring dockerfile: 460B                                                                                                                                                                              0.0s
 => [internal] load .dockerignore                                                                                                                                                                                 0.0s
 => => transferring context: 2B                                                                                                                                                                                   0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-buster                                                                                                                                      2.0s
 => CACHED [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:830e161433edfe047a23ebc99c12ee0eb1dc0a50e6b5f1c98e869ac271786632                                                                    0.0s
 => [internal] load build context                                                                                                                                                                                 0.1s
 => => transferring context: 3.02kB                                                                                                                                                                               0.0s
 => ERROR [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto                                              67.8s
------
 > [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto:
12.90 Reading package lists...
17.05 Building dependency tree...
17.73 Reading state information...
19.10 The following additional packages will be installed:
19.10   binutils binutils-arm-linux-gnueabihf binutils-common bzip2 cpp cpp-8
19.10   dnsmasq-base dpkg-dev g++ g++-8 gcc gcc-8 git-man libasan5 libatomic1
19.10   libbinutils libbz2-1.0 libc-dev-bin libc6 libc6-dev libcap2 libcap2-bin
19.10   libcc1-0 libcurl3-gnutls libdbus-1-3 libdpkg-perl libelf1 liberror-perl
19.10   libev4 libgcc-8-dev libgdbm-compat4 libgomp1 libgssapi-krb5-2 libidn11
19.10   libisl19 libk5crypto3 libkeyutils1 libkrb5-3 libkrb5support0 libldap-2.4-2
19.10   libldap-common libmnl0 libmpc3 libmpfr6 libnetfilter-conntrack3
19.11   libnfnetlink0 libnghttp2-14 libnl-3-200 libnl-genl-3-200 libnl-route-3-200
19.11   libpcre2-8-0 libperl5.28 libpsl5 librtmp1 libsasl2-2 libsasl2-modules-db
19.11   libssh2-1 libssl1.1 libstdc++-8-dev libubsan1 libuv1 libwebsockets8 libwrap0
19.11   libxtables12 linux-libc-dev lsb-base make patch perl perl-modules-5.28
19.11   xz-utils
19.13 Suggested packages:
19.13   binutils-doc bzip2-doc cpp-doc gcc-8-locales resolvconf debian-keyring
19.13   gcc-8-doc libstdc++6-8-dbg gcc-multilib manpages-dev autoconf automake
19.13   libtool flex bison gdb gcc-doc libgcc1-dbg libgomp1-dbg libitm1-dbg
19.13   libatomic1-dbg libasan5-dbg liblsan0-dbg libtsan0-dbg libubsan1-dbg
19.13   libmpx2-dbg libquadmath0-dbg gettext-base git-daemon-run
19.13   | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs
19.13   git-mediawiki git-svn iproute2-doc glibc-doc libc-l10n locales gnupg
19.13   | gnupg2 sensible-utils bzr krb5-doc krb5-user libssl-doc libstdc++-8-doc
19.13   make-doc apparmor ed diffutils-doc perl-doc libterm-readline-gnu-perl
19.13   | libterm-readline-perl-perl libb-debug-perl liblocale-codes-perl
19.13 Recommended packages:
19.13   dns-root-data fakeroot gnupg | gnupg2 libalgorithm-merge-perl less
19.13   ssh-client libatm1 manpages manpages-dev libpam-cap dbus
19.13   libfile-fcntllock-perl liblocale-gettext-perl krb5-locales publicsuffix
19.13   libsasl2-modules
21.83 The following NEW packages will be installed:
21.83   binutils binutils-arm-linux-gnueabihf binutils-common build-essential bzip2
21.83   cpp cpp-8 dnsmasq dnsmasq-base dpkg-dev g++ g++-8 gcc gcc-8 git git-man
21.84   hostapd iproute2 libasan5 libatomic1 libbinutils libc-dev-bin libc6-dev
21.84   libcap2 libcap2-bin libcc1-0 libcurl3-gnutls libdbus-1-3 libdpkg-perl
21.84   libelf1 liberror-perl libev4 libgcc-8-dev libgdbm-compat4 libgomp1
21.84   libgssapi-krb5-2 libidn11 libisl19 libk5crypto3 libkeyutils1 libkrb5-3
21.84   libkrb5support0 libldap-2.4-2 libldap-common libmnl0 libmpc3 libmpfr6
21.84   libnetfilter-conntrack3 libnfnetlink0 libnghttp2-14 libnl-3-200
21.84   libnl-genl-3-200 libnl-route-3-200 libpcre2-8-0 libperl5.28 libpsl5 librtmp1
[... cut ...]
   3 | >>> RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
   4 |
   5 |     FROM base AS python-deps
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto" did not complete successfully: exit code: 100
Failed to build Docker image, stopping script
pi@raspberrypi:~/tuya-cloudcutter $`

Cossid, 2023-10-29:
Those errors are all timeouts, which is issues with your network, not docker or any included script.  Adjust network settings as necessary and try again.

amedeodivito, 2023-10-29:
hi, thanks for the reply;
I had the Ethernet network disabled, I enabled it and reinstalled everything on the SD
unfortunately it gives the same error
502 connection timed out [IP: 151.101.242.132 80]
I'm not very familiar with Linux but I'm sure I followed the procedures described exactly.
I tell you that a few days ago everything went ok and I unlocked a Tuya device, then by mistake I broke the SD card
If you have any other suggestions I would be grateful.
Amedeo.
(I use google translator)

amedeodivito, 2023-10-30:
my network works, I don't understand the reason for those timeouts