tuya-cloudcutter/tuya-cloudcutter #364: [BK7231N] [CBU] [DUMP] Tuya Generic Single Phase Din Rail Wifi Intelligent Energy monitoring switch ZMAi-90

ataa, 2023-04-21:
[image]

[ZMAi-90-Slow-readResult_BK7231N_QIO_2023-21-4--04-02-38.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11295240/ZMAi-90-Slow-readResult_BK7231N_QIO_2023-21-4--04-02-38.zip)

Cossid, 2023-04-21:
The attached dump is mostly (but not completely) null bytes and is therefor invalid.

ataa, 2023-04-21:
What could be the reason for that? here is a flash read log I got success message for flash read and CRC check:

```
Getting bus success!
Going to set baud rate setting (115200)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... Ok! Reading 0x1000... Ok! Reading 0x2000... Ok! Reading 0x3000... Ok! Reading 0x4000... Ok! Reading 0x5000... Ok! Reading 0x6000... Ok! Reading 0x7000... Ok! Reading 0x8000... Ok! Reading 0x9000... Ok! Reading 0xA000... Ok! Reading 0xB000... Ok! Reading 0xC000... Ok! Reading 0xD000... Ok! Reading 0xE000... Ok! Reading 0xF000... Ok! Reading 0x10000... Ok! Reading 0x11000... Ok! Reading 0x12000... Ok! Reading 0x13000... Ok! Reading 0x14000... Ok! Reading 0x15000... Ok! Reading 0x16000... Ok! Reading 0x17000... Ok! Reading 0x18000... Ok! Reading 0x19000... Ok! Reading 0x1A000... Ok! Reading 0x1B000... Ok! Reading 0x1C000... Ok! Reading 0x1D000... Ok! Reading 0x1E000... Ok! Reading 0x1F000... Ok! Reading 0x20000... Ok! Reading 0x21000... Ok! Reading 0x22000... Ok! Reading 0x23000... Ok! Reading 0x24000... Ok! Reading 0x25000... Ok! Reading 0x26000... Ok! Reading 0x27000... Ok! Reading 0x28000... Ok! Reading 0x29000... Ok! Reading 0x2A000... Ok! Reading 0x2B000... Ok! Reading 0x2C000... Ok! Reading 0x2D000... Ok! Reading 0x2E000... Ok! Reading 0x2F000... Ok! Reading 0x30000... Ok! Reading 0x31000... Ok! Reading 0x32000... Ok! Reading 0x33000... Ok! Reading 0x34000... Ok! Reading 0x35000... Ok! Reading 0x36000... Ok! Reading 0x37000... Ok! Reading 0x38000... Ok! Reading 0x39000... Ok! Reading 0x3A000... Ok! Reading 0x3B000... Ok! Reading 0x3C000... Ok! Reading 0x3D000... Ok! Reading 0x3E000... Ok! Reading 0x3F000... Ok! Reading 0x40000... Ok! Reading 0x41000... Ok! Reading 0x42000... Ok! Reading 0x43000... Ok! Reading 0x44000... Ok! Reading 0x45000... Ok! Reading 0x46000... Ok! Reading 0x47000... Ok! Reading 0x48000... Ok! Reading 0x49000... Ok! Reading 0x4A000... Ok! Reading 0x4B000... Ok! Reading 0x4C000... Ok! Reading 0x4D000... Ok! Reading 0x4E000... Ok! Reading 0x4F000... Ok! Reading 0x50000... Ok! Reading 0x51000... Ok! Reading 0x52000... Ok! Reading 0x53000... Ok! Reading 0x54000... Ok! Reading 0x55000... Ok! Reading 0x56000... Ok! Reading 0x57000... Ok! Reading 0x58000... Ok! Reading 0x59000... Ok! Reading 0x5A000... Ok! Reading 0x5B000... Ok! Reading 0x5C000... Ok! Reading 0x5D000... Ok! Reading 0x5E000... Ok! Reading 0x5F000... Ok! Reading 0x60000... Ok! Reading 0x61000... Ok! Reading 0x62000... Ok! Reading 0x63000... Ok! Reading 0x64000... Ok! Reading 0x65000... Ok! Reading 0x66000... Ok! Reading 0x67000... Ok! Reading 0x68000... Ok! Reading 0x69000... Ok! Reading 0x6A000... Ok! Reading 0x6B000... Ok! Reading 0x6C000... Ok! Reading 0x6D000... Ok! Reading 0x6E000... Ok! Reading 0x6F000... Ok! Reading 0x70000... Ok! Reading 0x71000... Ok! Reading 0x72000... Ok! Reading 0x73000... Ok! Reading 0x74000... Ok! Reading 0x75000... Ok! Reading 0x76000... Ok! Reading 0x77000... Ok! Reading 0x78000... Ok! Reading 0x79000... Ok! Reading 0x7A000... Ok! Reading 0x7B000... Ok! Reading 0x7C000... Ok! Reading 0x7D000... Ok! Reading 0x7E000... Ok! Reading 0x7F000... Ok! Reading 0x80000... Ok! Reading 0x81000... Ok! Reading 0x82000... Ok! Reading 0x83000... Ok! Reading 0x84000... Ok! Reading 0x85000... Ok! Reading 0x86000... Ok! Reading 0x87000... Ok! Reading 0x88000... Ok! Reading 0x89000... Ok! Reading 0x8A000... Ok! Reading 0x8B000... Ok! Reading 0x8C000... Ok! Reading 0x8D000... Ok! Reading 0x8E000... Ok! Reading 0x8F000... Ok! Reading 0x90000... Ok! Reading 0x91000... Ok! Reading 0x92000... Ok! Reading 0x93000... Ok! Reading 0x94000... Ok! Reading 0x95000... Ok! Reading 0x96000... Ok! Reading 0x97000... Ok! Reading 0x98000... Ok! Reading 0x99000... Ok! Reading 0x9A000... Ok! Reading 0x9B000... Ok! Reading 0x9C000... Ok! Reading 0x9D000... Ok! Reading 0x9E000... Ok! Reading 0x9F000... Ok! Reading 0xA0000... Ok! Reading 0xA1000... Ok! Reading 0xA2000... Ok! Reading 0xA3000... Ok! Reading 0xA4000... Ok! Reading 0xA5000... Ok! Reading 0xA6000... Ok! Reading 0xA7000... Ok! Reading 0xA8000... Ok! Reading 0xA9000... Ok! Reading 0xAA000... Ok! Reading 0xAB000... Ok! Reading 0xAC000... Ok! Reading 0xAD000... Ok! Reading 0xAE000... Ok! Reading 0xAF000... Ok! Reading 0xB0000... Ok! Reading 0xB1000... Ok! Reading 0xB2000... Ok! Reading 0xB3000... Ok! Reading 0xB4000... Ok! Reading 0xB5000... Ok! Reading 0xB6000... Ok! Reading 0xB7000... Ok! Reading 0xB8000... Ok! Reading 0xB9000... Ok! Reading 0xBA000... Ok! Reading 0xBB000... Ok! Reading 0xBC000... Ok! Reading 0xBD000... Ok! Reading 0xBE000... Ok! Reading 0xBF000... Ok! Reading 0xC0000... Ok! Reading 0xC1000... Ok! Reading 0xC2000... Ok! Reading 0xC3000... Ok! Reading 0xC4000... Ok! Reading 0xC5000... Ok! Reading 0xC6000... Ok! Reading 0xC7000... Ok! Reading 0xC8000... Ok! Reading 0xC9000... Ok! Reading 0xCA000... Ok! Reading 0xCB000... Ok! Reading 0xCC000... Ok! Reading 0xCD000... Ok! Reading 0xCE000... Ok! Reading 0xCF000... Ok! Reading 0xD0000... Ok! Reading 0xD1000... Ok! Readi [... cut ...] 0... Ok! Reading 0x1F6000... Ok! Reading 0x1F7000... Ok! Reading 0x1F8000... Ok! Reading 0x1F9000... Ok! Reading 0x1FA000... Ok! Reading 0x1FB000... Ok! Reading 0x1FC000... Ok! Reading 0x1FD000... Ok! Reading 0x1FE000... Ok! Reading 0x1FF000... Ok! 
Basic read operation finished, but now it's time to verify...
Starting CRC check for 512 sectors, starting at offset 00
CRC matches 0x81C88304!
All read!
Loaded total 0x200000 bytes 
Wrote 2097152 to readResult_BK7231N_QIO_2023-21-4--04-02-38.bin
```

Cossid, 2023-04-21:
I can't say what the issue is, but I'm assuming CRC passes because the CRC for all null bytes is also null bytes.  It appears to be all 0xFF data until the storage near the very end.

ataa, 2023-04-21:
@kuba2k2 Unfortunately, after backing up the flash, I flashed it with OpenBeken firmware as I thought I have the factory firmware. I'll try with another one that I have, de-soldering the module is a pain as it is too close to the LCD and since this switch has tuya mcu, I wasn't able to flash/read it without de-soldering the module.

ataa, 2023-04-21:
There's an external flash (24C02H) on the board, connected to P15 and P17 of CBU module, Maybe the firmware is on the flash and the internal flash is free!

kuba2k2, 2023-04-21:
24C02 is an EEPROM, not a flash.

Cossid, 2023-04-21:
So #162 has the app partition with no storage, so we couldn't verify a schema, and you had storage with no app.  Combined, it looks like we're finally able to get a profile together, added as `Tuya Generic Zmai-90 Digital Energy Metering Switch`

No configuration available, but it appears to be TuyaMCU driven.

ataa, 2023-04-21:
Tried again on a second device.
 
ROM dump and Flash dump:
[ZMAi-90.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11297641/ZMAi-90.zip)

Cossid, 2023-04-21:
Yep, that is a complete and valid dump, and I have verified it matches our combined bin output exactly.

ataa, 2023-04-21:
So, There's no config even in this full dump. right?

Cossid, 2023-04-21:
Correct.  There is no user_param_key section present, which usually means it's all driven by TuyaMCU on RX1/TX1 and DPIDs.