tuya-cloudcutter/tuya-cloudcutter #365: Scanning for AP using a USB WiFi adapter on Raspberry Pi 2B

I3lackshirts, 2023-04-24:
I have been going through the setup with a spare RPi 2B I had lying around. The issue I am having is I get stuck at searching for the AP and it cant find the device. I am using a usb wifi adapter that I believe is setup correctly as I am able to scan for networks using "sudo iwlist wlan0 scan". Going through the closed issues I tried a few different things regarding the Network Manager, nmcli radio, and the following commands:

```

pi@pi:~/tuya-cloudcutter $ sudo systemctl disable dhcpcd
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable dhcpcd
pi@pi:~/tuya-cloudcutter $ sudo systemctl stop dhcpcd
pi@pi:~/tuya-cloudcutter $ nmcli radio wifi on
Error: failed to set Wi-Fi radio: Not authorized to perform this operation
pi@pi:~/tuya-cloudcutter $ cd ..
pi@pi:~ $ nmcli radio wifi on
Error: failed to set Wi-Fi radio: Not authorized to perform this operation

```

I am unsure what else I can do and I am wondering if I can even use a wifi adapter with a RPi2? Any other ideas? I am getting the nightlight in the slow blink state and can see the AP on my phone as "Globe Suite-613E" Some background as well is I have connected in to the tuya app in the past and tried flashing over USB. Don't believe anything is wrong but not sure if a factory rest is needed or it disconnected from Tuya app. Currently I have it running in Tuya but want to convert it to ESPHome.

Below is the rest of my commands as I step through the docker:

```
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: globe-motion-esphome.bin
 > globe-motion-esphome.bin

[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Globe Electric
   FCmila
   FLSNT
   Feit
   Filohome
   Fitop
   Geeni
 > Globe Electric
   Globo Lighting
   Gosund
   Hama
   Helloify
   Hihome
   Hombli

[?] Select the article number of your device: 50239 Motion Night Light
   35804 RGB E26 LED Bulb
   37783 RGBCT Bulb v2.0.0
   37783 RGBCT Bulb v2.0.3
   50151 Smart 2-Outlet Outdoor Plug
 > 50239 Motion Night Light
   50323 RGBCT Recessed Light
   50329 Smart Plug
   50347 Smart-Plug

[?] Select the firmware version and name: 1.3.2 - BK7231T / oem_bk7231s_light_pir_globe
 > 1.3.2 - BK7231T / oem_bk7231s_light_pir_globe

Selected Device Slug: globe-electric-50239-motion-night-light
Selected Profile: oem-bk7231s-light-pir-globe-1.3.2-sdk-1.0.2-40.00
Selected Firmware: /work/custom-firmware/globe-motion-esphome.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
..........................................................................................................................................................................................................................................................................
..........................................................................................................................................................................................................................................................................
..........................................................................................................................................................................................................................................................................
..........................................................................................................................................................................................................................................................................
```

Cossid, 2023-04-24:
> pi@pi:\~/tuya-cloudcutter $ nmcli radio wifi on
> Error: failed to set Wi-Fi radio: Not authorized to perform this operation
> pi@pi:\~/tuya-cloudcutter $ cd ..
> pi@pi:\~ $ nmcli radio wifi on
> Error: failed to set Wi-Fi radio: Not authorized to perform this operation

Doesn't really show if nmcli is working correctly or not, you probably need to run them with sudo as well.

My guess is that NetworkManager still isn't correctly installed/running.  See https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md for specific instructions.  You may need to use the specific version of Raspbian OS listed there.

You may also need to look into the `-r` and `-w` flags.  See `./tuya-cloudcutter.sh -h` for parameter descriptions.

Cossid, 2023-05-04:
Closing as stale.  Feel free to re-open this issue if you are able to supply more information.

closed 2023-05-04 by Cossid

felipecrs, 2024-11-30:
I was getting the same issue. Fixed it by using `sudo` to run `./tuya-cloudcutter.sh` (I was trying to run it without `sudo` for some reason).