tuya-cloudcutter/tuya-cloudcutter #366: Cannot cloudcut/flash Aubess Plug 20A

czerwony03, 2023-04-24:
Hi,
I'm trying to test tuya-cloudcutter with Aubess Plug, but I cannot get it working.
I'm starting script, putting plug into AP mode, exploit ok, putting into AP mode again, device is being configured.... but than... It looks like device is connecting to laptop AP, but it's only querying some tuya domain, and not sending any requests requesting flash or anything....

Here is my log from custom flash attempt:
`$ sudo ./tuya-cloudcutter.sh -v
Building cloudcutter docker image
[+] Building 1.5s (13/13) FINISHED                                              
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 460B                                       0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-bus  1.4s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:83  0.0s
 => [internal] load build context                                          0.0s
 => => transferring context: 3.02kB                                        0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv              0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                        0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                   0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipe  0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                 0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                  0.0s
 => exporting to image                                                     0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:b84d1d1772f8305876e0e06147659f294522a8362d289  0.0s
 => => naming to docker.io/library/cloudcutter                             0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: OpenBK7231N_UG_1.17.30.bin
 > OpenBK7231N_UG_1.17.30.bin

[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Aubess
   Antela
   Aoycocr
   Arlec
   Atarm
   Athom
   Atomi
 > Aubess
   Avatar
   Avatto
   BNETA IoT
   Baytion
   Beca
   Biltema

[?] Select the article number of your device: 20A Plug
   16A Mini Smart Switch
 > 20A Plug
   IR Remote Controller

[?] Select the firmware version and name: 1.1.8 - BK7231N / oem_bk7231n_plug
 > 1.1.8 - BK7231N / oem_bk7231n_plug

Selected Device Slug: aubess-20a-plug
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00
Selected Firmware: /work/custom-firmware/OpenBK7231N_UG_1.17.30.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
[... 352 lines cut ...]
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 DHCPREQUEST(wlp2s0) 10.42.42.31 1c:90:ff:26:ed:f6 
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 tags: wlp2s0
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 DHCPACK(wlp2s0) 10.42.42.31 1c:90:ff:26:ed:f6 wlan0
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  5
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Apr 24 21:33:53 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
`

I will be grateful for any help.

Best regards

Cossid, 2023-04-24:
As far as I'm aware, this can only happen due to poor wifi signal.  Make sure your cloudcutter and target device as as close to each other as possible.

The device is properly connected to the correct wifi, gets a dhcp signal (though clearly struggles to the first time, another sign of poor signal), then makes a dns request, and cloudcutter sends a response, then the device doesn't make any subsequent requests, which means it's having connection issues.

The fact that it is making new dhcp requests minutes after failing DNS is further signs of a poor connection and the device attempting to reconnect.

czerwony03, 2023-04-24:
Laptop was just beside socket. I have tried second PC but with same results. Maybe socket  is broken....

Cossid, 2023-04-24:
Hard to say.  All I can confirm is that CloudCutter is doing everything correctly, and the device is just having general network connection issues.  The amount of time between DHCP request/ack and first DNS request is unusually high.

Cossid, 2023-05-04:
Closing as stale, as this doesn't appear to directly be a CloudCutter issue.  Feel free to re-open this issue if you are able to supply more information.

closed 2023-05-04 by Cossid