tuya-cloudcutter/tuya-cloudcutter #367: Ledvance AC31267 outdoor plug request

sebagsm, 2023-04-24:


Cossid, 2023-04-24:
We can't add support for a device without a dump of the firmware.  If you are able to provide one, support can be added.  If not, you may try to find a compatible firmware by finding which firmware version your device has: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

sebagsm, 2023-04-25:
on it :) 

[image]

Cossid, 2023-05-04:
@sebagsm No dump has been supplied yet, so I'm closing this issue as stale.  If you are able to provide a firmware dump, please attach it here and re-open this issue.

mbagge, 2024-09-30:
Dump added. I hope it has the right format.
[ltchiptool_bk72xx_2024-09-26_16-33-17.zip](https://github.com/user-attachments/files/17190974/ltchiptool_bk72xx_2024-09-26_16-33-17.zip)

Cossid, 2024-09-30:
Added as `LEDVANCE SmartPlus AC31267 Outdoor Plug v1.1.8`

closed 2024-09-30 by Cossid