tuya-cloudcutter/tuya-cloudcutter #368: Firmware 1.1.15?

sining951, 2023-04-26:
I have a generic tuya plug that I just got that seems to have new firmware 1.1.15. Is this supported somehow as I don't see it anywhere in the list of devices/firmware. 

The older versions of this plug used to have 1.1.8 and I did successfully flash openbeken on them before but now when I try this on the new plug, it tells me 'the profile you selected did not result in a successful exploit'

Cossid, 2023-04-26:
There is currently no 1.1.15 firmware supported by CloudCutter.  This would require a serial dump to support currently.  Feel free to re-open this issue if you are able to provide a firmware.

closed 2023-04-26 by Cossid

dante1613, 2023-04-27:
Hi i have dump with this firmware v1.1.15

Cossid, 2023-04-27:
@dante1613 Yes, you can zip it and attach it to this issue.

dante1613, 2023-04-27:
Since I did not know which dump would be better for you, I made two of them. One via windows and bk7231flasher_1.0.8 and the second via bk7231tools and debian 11.

Each archive has 2 versions, clean (before connected) and paired to tuya through my home network
[DUMPS.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11347461/DUMPS.zip)

kuba2k2, 2023-04-27:
FYI, both of these tools produce the exact same formats of dumps, so it's just a matter of personal preference which one you use.

dante1613, 2023-04-27:
Great, then I will use windows and bk7231flasher, since I am far from Linux

Cossid, 2023-04-27:
Unfortunately, Smart Plug firmware 1.1.15 is patched and not vulnerable to CloudCutter.  Serial flashing will be your only option.

dante1613, 2023-04-27:
It’s sad, of course, I thought it would be useful for the rest, because I don’t want to disassemble the plug anymore if I have to buy another one