tuya-cloudcutter/tuya-cloudcutter #376: [BK7231N] [CBU] [DUMP] Gvvoo Rechargable RGB Atmospher Light HCP20515 (Wifi/BLE)

ataa, 2023-04-30:
[image]

ROM Dump is also included.
[Rechargable-RGB-bowl-light-ble-wifi-N-CBU.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11361061/Rechargable-RGB-bowl-light-ble-wifi-N-CBU.zip)

Cossid, 2023-04-30:
Added as `Tuya Generic HCP20515 Rechargable RGB Atmosphere Light`.  Tuya-Generic because there is no branding on the box, and a quick search shows it for sale under at least 3 vendor/brands.

No device configuration, likely a TuyaMCU device.

closed 2023-04-30 by Cossid

ataa, 2023-04-30:
- Gvvoo is the name of the app to control the light.
- Device does not start the AP on boot, instead I can see a new bluetooth device ready to pair with Gvvoo app.
- There's no MCU on the pcb, everything is connected to the CPU module (PWM, IR, MIC, Battery charger)
- I did not pair it with the app, configuration might be done by the app via bluetooth.
- So far I managed to find these pins:

 "pins": {
    "6": "PWM;0",
    "7": "PWM;1",
    "8": "PWM;2",
    "23": "ADC;5",
    "28": "Btn_SmartLED;3"
  },

P20: Charger connected (?)
P26: IR (but there's no receiver on the pcb)