tuya-cloudcutter/tuya-cloudcutter #377: ELEGRP Dimmer Dump

efirestone, 2023-05-01:
Attached is the dump for the [ELEGRP Wi-Fi Touch Dimmer Switch](https://www.amazon.com/dp/B0BHQWTBWB). I'm interested in using CloudCutter with it if someone would be kind enough to add a profile for it. Thank you!

[dump_elegrp-dimmer.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11363601/dump_elegrp-dimmer.bin.zip)

Cossid, 2023-05-01:
Added as `ELEGRP DPR10 Touch Dimmer Switch`.  No device configuration available, appears to be TuyaMCU controlled.

closed 2023-05-01 by Cossid

efirestone, 2023-05-01:
Amazing, thank you!

For the "no device configuration available" part, I know you can set the minimum dim level in the Smart Life or ELEGRP app (not sure which). Is there any information I can gather to get those exposed in the profile?

openshwprojects, 2023-05-10:
@efirestone  you can use TuyaMCU analyzer that will automaticaly show Tuya packet content:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20528459#20528459

razzamatazm, 2023-09-14:
I'm having a hard time getting my switch into AP Mode. Do you have any tips?

razzamatazm, 2023-09-14:
I've tried all of these methods. Turning on and off, holding the power button. I can get it to fast blink by holding off button. Once it's fast blinking, I try to do the same but it just keeps fast blinking until it times out. Losing my mind trying to get it to work.

Cossid, 2023-09-14:
Whatever puts it into fast-blink is the same that will put it into slow-blink.  You must start the progression from fast to slow relatively quickly, you can't let it sit in fast-blink mode.

Most devices you can have a short pause inbetween, but some you just have to do all 6x or 10x in one step instead of pausing between.

razzamatazm, 2023-09-14:
To put this switch in fast blink, you hold off for about 7 seconds. I've tried immediately holding again after it starts fast blinking multiple times now. I wonder if this is just a faulty switch.

Cossid, 2023-09-14:
Have you added this device to Smart Life?  Is it the same firmware version?  Do the instructions say to use the Smart Life app, or is there a brand specific app like ELEGRP?  If they instruct their own app, they might not have included slow blink mode, Cloudy Bay switches are like this.  However, they usually use a custom firmware when they do that, but this one appears to be a bog-standard TuyaMCU firmware, not a brand-specific one.

Generally, the rule of thumb is hold the button down until it starts changing states, let go and verify it is fast blinking then immediately do that again.  If it goes right back into fast blink mode, it is possible it doesn't have a slow blink mode.

kuba2k2, 2023-09-14:
If you manage to flash it with Cloudcutter - you can use the latest version of ltchiptool to generate ESPHome YAML config for your device.

<details>
<summary>Screenshot</summary>

[image]

</details>

The latest versions support generating for TuyaMCU devices.

razzamatazm, 2023-09-14:
I utilized the Smart Life app to setup the switch at home, but they also
have their own app. I'll do a little digging on firmware. I'll be bummed if
it's been patched out.

efirestone, 2023-09-24:
This is the ESPHome config I've been using based on my own testing/reverse engineering:
```
substitutions:
  device_name: some_room_light
  friendly_name: Some Room Light
  name: "touchdimmer1"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

libretiny:
  board: cb2s
  framework:
    version: dev

# Enable logging
# Make sure logging is not using the serial port
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: !secret ha_encryption_key

# Over the air updating
ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "${friendly_name}"
    password: !secret ap_password
  
  domain: .home

mdns:
  # Avoid network congestion since the hostname is configured
  # with a static IP on the router.
  # Your config may not have this and this should be left enabled.
  disabled: True

# https://esphome.io/components/tuya.html
uart:
  # https://docs.libretiny.eu/boards/cb2s/
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

# Datapoint descriptions available here:
# https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/elegrp-dtr10-single-pole-touch-dimmer-switch.json

light:
  - platform: "tuya"
    name: ${friendly_name}
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 10
    max_value: 1000

  # The low end trim is the minimum allowed brightness
  # If the attached light is flickering, or not turning on when
  # at the lowest setting, then increase the low end trim so
  # that the dimmer doesn't dim to as low of a value.
  # When this switch is turned on then the adjustment mode is
  # engaged and will show the current low end trim value.
  # Tapping the physical switch while it's in this mode will return it to
  # its last-known normal setting and exit low end trim edit mode.
  - platform: "tuya"
    name: Low End Trim
    entity_category: config
    dimmer_datapoint: 3
    switch_datapoint: 114
    min_value: 0
    max_value: 500

number:
  - platform: "tuya"
    name: "Brightness"
    entity_category: config
    number_datapoint: 2
    min_value: 10
    max_value: 1000
    step: 1
    mode: slider

  - platform: "tuya"
    name: "Night Indicator Light Brightness"
    entity_category: config
    number_datapoint: 101
    unit_of_measurement: '%'
    min_value: 0
    max_value: 100
    step: 1
    mode: slider

  - platform: "tuya"
    name: "Unknown 5"
    entity_category: config
    number_datapoint: 5
    min_value: 550
    max_value: 1000
    step: 1
    mode: slider

  - platform: "tuya"
    name: "Unknown 108"
    entity_category: config
    number_datapoint: 108
    min_value: 10
    max_value: 1000
    step: 1
    mode: slider

  - platform: "tuya"
    name: "Unknown 111"
    entity_category: config
    number_datapoint: 111
    min_value: 0
    max_value: 86399
    step: 1
    mode: slider

  # Set this to a value in seconds and the light will
  # turn off when that time expires.
  - platform: "tuya"
    name: "Countdown Timer"
    entity_category: config
    number_datapoint: 6
    unit_of_measurement: 's'
    min_value: 0
    max_value: 86399
    step: 1

select:
  - platform: "tuya"
    name: "Bulb Type"
    entity_category: config
    enum_datapoint: 4
    optimistic: true
    options:
      0: CFL
      1: Incandescent
      2: Halogen
      3: LED

  - platform: "tuya"
    name: "Fade On Speed"
    entity_category: config
    enum_datapoint: 103
    optimistic: true
    options:
      0: Immediate
      1: 1 Second
      2: 5 Seconds
      3: 15 Seconds

  - platform: "tuya"
    name: "Fade Off Speed"
    entity_category: config
    enum_datapoint: 104
    optimistic: true
    options:
      0: Immediate
      1: 1 Second
      2: 5 Seconds
      3: 15 Seconds

sensor:
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

switch:
  - platform: "tuya"
    name: "Switch 110"
    entity_category: config
    switch_datapoint: 110
```