tuya-cloudcutter/tuya-cloudcutter #382: Door sensor CB3S no MCU SmartLED 8072

openshwprojects, 2023-05-05:
SmartLED 8072:
[image]
That one should be interesting
[CB3S_Door_Sensor_NoMCU.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11406087/CB3S_Door_Sensor_NoMCU.zip)

Cossid, 2023-05-05:
Not sure on this one.  It doesn't appear to have any storage for configuration.  It has a custom firmware that is specific to this device, and the SDK version is one previously unseen and unexpected.  It looks to share a lot with SDK 2.3.1 (based on versions of components), but it doesn't share the same exploit pattern.

It also has a compile date of Aug 2022, which means there is a chance it it patched, but that's not verified yet either, I can't seem to find the code that would verify that yet.

openshwprojects, 2023-05-05:
All I can say is that's not a TuyaMCU device and it has a LED, a door sensor and a buzzer on GPIO, and it works with OBK well.

Cossid, 2023-05-18:
@openshwprojects Do you happen to have a mapping for the GPIO?  This device appears to default to a long sleep mode (12 minutes at a time), and I need to know which GPIO puts it into pairing mode to test out supporting it in CloudCutter.

openshwprojects, 2023-05-18:
@Cossid  Of course! Normally I'd ask you to wait for the upcoming review guide/article, but since you're a dev, here's a sneak peak:

P7 (PWM1)  - RED LED
P9 (PWM3) - Buzzer
P6 - door sensor no pullup
DSEdge 0

Cossid, 2023-05-18:
@openshwprojects No button?  How does this device enter pairing mode?

openshwprojects, 2023-05-18:
Here's a button spoiler for you:
[image]

openshwprojects, 2023-05-18:
And some other photos if you want to double-check:
[image]
[image]
[image]
[image]
[image]

Cossid, 2023-05-18:
Looks like that button is probably on P26.  When I tried grounding P26, the device just boot looped, but I wasn't specifically on a CB3S board, not sure if that makes a difference.

Cossid, 2023-05-18:
> EDIT: Btw, are you sure it's 12 minutes long sleep? I'd rather expect it to use GPIO interrupt to wake up on door event. In my opinion, it is not optimal to sleep just for 12 minutes, maybe it wakes up to check smth/report self state, but still.. 12 minutes seems like a very short time.

I'm sure it wakes on GPIO input as well (I don't have the actual device to test, but that is the usual pattern), but the core device sleeps for 12 minutes between sending idle status updates by default.

openshwprojects, 2023-05-18:
I am not sure what could cause boot loop, but here are my two observations:

1. When I tried to swap 2MB flash dumps with OpenBeken between two CB3S modules, both of them were experiencing very strange and serious WiFi issues, they were basically unusable. I only managed to get one of them in pairing mode. It was that way not because of the OBK config, which I cleared, but because of RF partition/something like that. I haven't investigated it more, but suggests that there some kind of chip-specific calibration data in the flash memory.
Do you know anything about that, have you tried running 2MB dumps from other people? It may be OBK-specific problem, I don't know yet.

2. I haven't checked it for this particular device, but battery powered devices are also using ADC3 pin (P23) to measure battery voltage. You have most likely seen battery data in user_param_keys extracted from other devices.
Maybe you have flashed dump on device where ADC3 hangs in air and it somehow causes firmware to think that battery is too low?

openshwprojects, 2023-05-18:
3. isn't that button, strangely enough, connected to VDD and not GND?
[image]

Cossid, 2023-05-18:
> 1. When I tried to swap 2MB flash dumps with OpenBeken between two CB3S modules, both of them were experiencing very strange and serious WiFi issues, they were basically unusable. I only managed to [...]

I flash full firmwares onto my test board, a CB8P all the time, and they almost always work.  The exceptions are always battery powered devices.  I'm not sure about the storage, I thought it was just a mac address stored, I don't know anything beyond that.

> 2. I haven't checked it for this particular device, but battery powered devices are also using ADC3 pin (P23) to measure battery voltage. You have most likely seen battery data in user_param_keys [...]

Possible.  I generally only cross-flash when I specifically need to test a new exploit offset/pattern, and up until now I've never had to do that with a battery powered device, they've all been compatible with other patterened firmwares.

> 3. isn't that button, strangely enough, connected to VDD and not GND?

I'm not great on the EE side, but yeah, it kinda looks like it, guess I probably need to pull it high instead of low then.  I'll give that a try later.