tuya-cloudcutter/tuya-cloudcutter #386: tuya generic lspa9 plug FW 1.1.15 do not work

catalinbordan, 2023-05-11:
Hello,
I have bought 3 plugs tuya generic lspa9 with firmware 1.1.15(main module V1.1.15, other type V1.0.5) for which it will not work to run the tuya-cloudcutter. It connects to the AP of the plug, but it says it does not find expected IP. try 3 times and then stop.
One of the plug I have disassembled and I have found the chip CB2S [BK7231N]. 
I have used my iron "skills" to get 4 fire to use usb to serial, but no succes to flash the chip with openbeken tool.
With a big battle I have succeeded to take out the chip from the plug board, then backup the firmware and flash with openbeken.
But unfortunately my skills almost destroyed the tracks of the circuits. I was forced to use some wires and still no success. I can not calibrate the plug, in idle I have 24W, the physical button from the plug does not work and in the logs it seems the button is long pressed("Button_OnLongPressHold").

My question: is it possible now for the rest of the plugs to use tuya-cloudcutter if I have the original firmware dump and some settings of the pins?
here it is:
{
	"sel_pin_pin":"24",
	"rl1_lv":"1",
	"bt1_pin":"10",
	"net_trig":"4",
	"jv":"1.0.5",
	"netled1_lv":"0",
	"netled_reuse":"0",
	"bt1_type":"0",
	"ffc_select":"0",
	"nety_led":"1",
	"vi_pin":"6",
	"resistor":"1",
	"over_cur":"30000",
	"bt1_lv":"0",
	"reset_t":"5",
	"netled1_pin":"8",
	"chip_type":"0",
	"lose_vol":"90",
	"over_vol":"265",
	"module":"CB2S",
	"ele_pin":"7",
	"ch_cddpid1":"9",
	"ch1_stat":"2",
	"rl1_type":"0",
	"ch_num":"1",
	"ele_fun_en":"1",
	"rl1_pin":"26",
	"netn_led":"0",
	"vol_def":"0",
	"ch_dpid1":"1",
	"sel_pin_lv":"1",
	"crc":"101"
}

Device configuration, as extracted from Tuya: 
- BL0937 SEL on P24
- Button (channel 1) on P10
- BL0937 VI on P6
- WiFi LED on P8
- BL0937 ELE on P7
- Relay (channel 1) on P26
[readResult_BK7231N_QIO_2023-06-5--21-45-07.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11453406/readResult_BK7231N_QIO_2023-06-5--21-45-07.zip)

zoidy, 2023-05-11:
It seems like 1.1.15 might be [unexploitable](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/368)

Cossid, 2023-05-11:
Correct, BK7231N Plug v1.1.15 is patched and cannot be exploited by CloudCutter.  Serial flashing will be your only option.

closed 2023-05-11 by Cossid