tuya-cloudcutter/tuya-cloudcutter #389: New Device: ELEGRP SPN10 Single Pole Switch

efirestone, 2023-05-14:


Cossid, 2023-05-14:
Added as `ELEGRP SPN10 Single Pole Switch`

closed 2023-05-14 by Cossid

efirestone, 2023-05-16:
Is there a way to tell if this device uses a Tuya MCU just from the dump? Opening it up is proving difficult, so before I destroy one I'm wondering if there's an easy way to tell from the dump. Things I know so far:
- It's a bk7231n since that's what worked for lightleak
- It's almost certainly a CB2S because the FCC ID is `2ANDL-CB2S`, which is the same as the ELEGRP STN10, which definitely has a CB2S (that one I've opened up). This also matches the FCC ID mentioned for the CB2S board [here](https://docs.libretiny.eu/boards/cb2s/)

If I flash it with ESPHome, I keep getting `Initialization failed at init_state 0`, which means I'm either using the wrong UART pins (unlikely since they work for the STN10; rx_pin: D4, tx_pin: D5), or the chip doesn't have a Tuya MCU, or...something else.

Any thoughts or help are appreciated.

efirestone, 2023-05-17:
Helpful as always. I wasn't aware of ESPHome-Kickstart. I have it installed and loaded the web interface successfully. Is there a way to view all pins at once, or scan the pins somehow? I went through every pin with every mode, but was only able to figure out one button: RX1 toggles as-expected when Mode=INPUT_PULLUP and you press the top of the physical toggle. I can't seem to figure out the bottom toggle button or anything else (guidance appreciated).

Also, what do the entries in [the schema](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/elegrp-spn10-single-pole-switch.json) mean if they're not TuyaMCU datapoint descriptions?

Cossid, 2023-05-17:
Not that I am aware of, you must do one pin at a time.  Does RX1 also toggle with Mode=INPUT_PULLUP when you press the bottom button, or did you not try because you had already found the top button on that pin?  If both are triggering RX1, you most likely have a TuyaMCU, and certainly do if they are already triggering the relay.

Those schema would map to TuyaMCU if there is one, but it's not a guarantee, that's just Tuya's internal format which can adapt to both TuyaMCU and GPIO driven devices, it's just their translation layer.

thirug010, 2024-01-31:
> Helpful as always. I wasn't aware of ESPHome-Kickstart. I have it installed and loaded the web interface successfully. Is there a way to view all pins at once, or scan the pins somehow? I went [...]

just in case if you did not find the template for the buttons this is template,
 Single Pole 1 Way:
 "pins": {
    "6": "Btn;1",
    "7": "Rel;1",
    "23": "Btn;1",
    "24": "LED_n;1"
  },

Single Pole 3Way ( with blue wire) 
 "pins": {
    "6": "Btn;1",
    "7": "Rel;1",
    "8": "dInput_NoPullUp_n;1",
    "23": "Btn;1",
    "24": "LED_n;1"
  },