tuya-cloudcutter/tuya-cloudcutter #390: [BK7231N] [CB2S] [DUMP] Tuya Generic 10A Wifi Breaker LSPA5BP V1.1

ataa, 2023-05-17:
[image]

[dump_lspa5-10a.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11502908/dump_lspa5-10a.zip)

Cossid, 2023-05-17:
Added as `Tuya Generic LSPA5BP 10A Breaker`

closed 2023-05-17 by Cossid

ataa, 2025-01-28:
I received some of the same breaker from the same seller, but was not able to flash it using cloudcutter, I managed to dump the firmware, How to check if it's patched?
[new-switch-bk72xx.zip](https://github.com/user-attachments/files/18573014/new-switch-bk72xx.zip)

Cossid, 2025-01-28:
> I received some of the same breaker from the same seller, but was not able to flash it using cloudcutter, I managed to dump the firmware, How to check if it's patched? [new-switch-bk72xx.zip](https://github.com/user-attachments/files/18573014/new-switch-bk72xx.zip)

That is plug firmware 1.1.17 which is already on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list.  Serial flashing will remain the only option for devices with that firmware.