tuya-cloudcutter/tuya-cloudcutter #394: DS101-JL SMART SWITCH 1 gang version dump

openshwprojects, 2023-05-18:
[readResult_BK7231N_QIO_DS101-JL_2023-13-5--18-29-06.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11508188/readResult_BK7231N_QIO_DS101-JL_2023-13-5--18-29-06.zip)
That one:
https://allegro.pl/oferta/dotykowy-wlacznik-swiatla-szklany-wifi-tuya-13127485395

kuba2k2, 2023-05-18:
Btw, I think you can actually post all of these dump packs in one issue. Not really an issue, but could help with building profiles in batch.

Cossid, 2023-05-18:
> Btw, I think you can actually post all of these dump packs in one issue. Not really an issue, but could help with building profiles in batch.

Separate issues is actually better for if we ever use the `github_issues` value we're setting.  It also helps me keep track of which ones have been completed.

Cossid, 2023-05-18:
Added as `Tuya Generic DS101-JL 1 Gang Smart Switch`

closed 2023-05-18 by Cossid