tuya-cloudcutter/tuya-cloudcutter #396: Merkury MI-BW210-999WW

naething, 2023-05-21:
I am getting the following message when I select the matching profile in cloudcutter:

[!] The profile you selected did not result in a successful exploit.

I think these bulbs may be a different version? The firmware from lightleak is attached.

[dump_merkury-MI-BW210-999WW.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11523682/dump_merkury-MI-BW210-999WW.zip)

Cossid, 2023-05-21:
That is indeed different, added as `Merkury Innovations MI-BW210-999WW RGBCT Bulb v2.9.16` and renamed the other to include it's version of 2.9.6

Interestingly, they have different LED drivers, the 2.9.6 has an sm2135, while your 2.9.16 has a bp5758d
See https://upk.libretiny.eu/ and select your device to see pin mappings.

closed 2023-05-21 by Cossid

naething, 2023-05-21:
That worked great! One slight issue... when I used the website you linked above the ESPHome YAML is not quite right.

If you look here:
https://github.com/esphome/esphome-docs/blob/current/components/output/bp5758d.rst

The bp5758 component is 1-indexed not 0-indexed, so all of the channels in the YAML need to be incremented by one.

Once I did that I was able to flash the light works as expected.

Cossid, 2023-05-22:
Yeah, that will eventually be fixed in UPK.

robertscheib, 2023-06-13:
Was the indexing corrected in UPK yet? If not, I'll manually make the modifications.

Cossid, 2023-06-13:
@robertscheib Yes, it has been corrected in UPK.