tuya-cloudcutter/tuya-cloudcutter #397: newer FEIT WiFi dimmer switch from Costco 06.Oct.2022

openshwprojects, 2023-05-21:
I am clearing up my old notes and private messages/emails and I've found this dump, described by user as newer FEIT WiFi dimmer switch from Costco
[newer FEIT WiFi dimmer switch Costco 06 Oct 2022.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11524423/newer.FEIT.WiFi.dimmer.switch.Costco.06.Oct.2022.zip)
It's a partial dump, so I don't know if it will be any use for you, so let me know if posting partial dumps here makes any sense.

Cossid, 2023-05-21:
Unfortunately, this is too incomplete, nothing I can do with this dump.  Doesn't even seem to extract the app partition correctly.  Closing this ticket since is unlikely you'll be able to get a corrected dump.

closed 2023-05-21 by Cossid

engineerAndyy, 2023-06-10:
Hello, can you use the following dump? I just took this dump today for the FEIT dimmer switch from Costco and saw this post when going to add a new issue. 
[ltchiptool_bk72xx_2023-06-10_12-12-29_Feit_Dimmer_SW.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11713314/ltchiptool_bk72xx_2023-06-10_12-12-29_Feit_Dimmer_SW.zip)

edwwardk, 2023-09-26:
not sure this is exactly the same device since i cant read the number on the chip but i got a full dump from the bk7231n version through soldered serial

https://www.feit.com/collections/smart-controls/products/smart-wi-fi-dimmer-3-pack-dimsmart-3-can

[feit-dimsmart-3-can-bk7231n.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12732079/feit-dimsmart-3-can-bk7231n.zip)

Cossid, 2023-09-27:
@edwwardk Yeah, it looks like this device ships with two different modules, one BK7231T, the other BK7231N (like yours).  We had a dump for a BK7231T model, but I have now added your N model as `Feit DIM-WIFI Dimmer Switch bk7231n v2.0.2` and renamed the other to `Feit DIM-WIFI Dimmer Switch bk7231t v1.1.71` to avoid confusion.

dasb00ter, 2024-11-07:
> @Cossid not sure this is exactly the same device since i cant read the number on the chip in the picture but i got a full dump from the bk7231n version from canadian costco recently through soldered serial
> 
> https://www.feit.com/collections/smart-controls/products/smart-wi-fi-dimmer-3-pack-dimsmart-3-can
> 
> [feit-dimsmart-3-can-bk7231n.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12732079/feit-dimsmart-3-can-bk7231n.zip)

@edwwardk was your N version using patched firmware? I was wondering if I can cloudcut these or do I have to serial flash. Also wondering if 3rd party firmware allowed control of the light in the toggle switch itself?

Cossid, 2024-11-07:
> @edwwardk was your N version using patched firmware? I was wondering if I can cloudcut these or do I have to serial flash. Also wondering if 3rd party firmware allowed control of the light in the [...]

As long as you can get it into AP mode, the 2.0.2 firmware on this device was vulnerable, hence a profile was added for it.
For TuyaMCU devices, the button presses are handled by the TuyaMCU itself, and should be unchanged regardless of what firmware is installed.  If you are referring to the LED status light, that appears exposed via DPID 101 as off/low/high (based on assumptions), but probably no control of the "brightness" led strip.

dasb00ter, 2024-11-07:
@Cossid. To clarify the led strip would function as it does with the stock firmware? Best guesses if the firmware displayed here is vulnerable. I'm guessing advice will be too try. 
[image]
[image]

Cossid, 2024-11-07:
Yes, the brightness strip would be unchanged by the installed firmware as well.

Unfortunately, 2.1.17 is [patched](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) unlike the 2.0.2 firmware.  You won't be able to use CloudCutter for that, only serial flashing.

dasb00ter, 2024-11-07:
Is there a pattern to the numbering for the firmware? I thought I didn't see it listed in the patched firmware list... Is it the last 3 digits?

Cossid, 2024-11-07:
It's the `Main Module` version, 2.1.17 has been on the list for a while now.