tuya-cloudcutter/tuya-cloudcutter #402: Merkury Smart Wi-Fi LED Strip MI-EW010-999WW

clutchdude, 2023-05-28:
This profile was already created but whenever I try to run the existing profile, I get a 

`[!] The profile you selected did not result in a successful exploit.`

Across two different computers. 

I pulled used the lightleak retrieval method(and updated some docs). I tried creating the profile as described but the tuya app never sent a multicast or the device never received it.

Either way, it'll be interesting to see if this matches the existing profile(meaning I've got device configuration issues) or it's a new version of it. 

[dump_merkury-led-strip.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11583881/dump_merkury-led-strip.zip)

Cossid, 2023-05-28:
Yep, the version we had was 0.0.3, the version you supplied was 0.0.2, and it looks like it doesn't allow you to upgrade from 0.0.2 to 0.0.3

I have added a new profile for 0.0.2 and renamed the prior to include 0.0.3 in the profile name.

closed 2023-05-28 by Cossid

clutchdude, 2023-05-28:
@Cossid  thank you for all the work on this project - I'm very impressed how quick and easy this was to overcome and make much easier for everyone else.