tuya-cloudcutter/tuya-cloudcutter #407: 15W SMART bulb RGB2700

corbolais, 2023-06-01:


Cossid, 2023-06-01:
Post a full log if you are able.

If it's requesting a DHCP IP address but doing nothing further, that's usually a sign of a poor connection.  Make sure the CloudCutter and target devices are as close to each other as possible, AP mode is not very powerful.

corbolais, 2023-06-01:
Thanks for coming back this quickly.

Distance: 1m. Unobstructed.

corbolais, 2023-06-01:
What part(s) of the log might be of interest? The full log is rather lengthy.

Cossid, 2023-06-01:
Length isn't an issue, just put it in code blocks here.
Anything after `wlan0: AP-ENABLED` would probably be useful.

corbolais, 2023-06-01:
```
wlan0: AP-ENABLED 
[14:15:23.281673 MQTT] Connected
Jun  1 14:15:31 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jun  1 14:15:31 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 DHCPDISCOVER(wlan0) MACADDR  
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 DHCPOFFER(wlan0) 10.42.42.34 MAC 
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  2
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jun  1 14:15:34 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 DHCPDISCOVER(wlan0) MACADDR 
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 DHCPOFFER(wlan0) 10.42.42.34 MACADDR 
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  2
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 DHCPREQUEST(wlan0) 10.42.42.34 MACADDR 
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 DHCPACK(wlan0) 10.42.42.34 MACADDR wlan0
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  5
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jun  1 14:15:35 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jun  1 14:15:52 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.34
Jun  1 14:15:52 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01

[14:15:53.397611 Log (Client)] Request: HTTPServerRequest(protocol='https', host='h3.iot-dns.com', method='POST', uri='/v2/url_config', version='HTTP/1.1', remote_ip='10.42.42.34')
[14:15:53.397655 LOG (Client)] ==== Request body ===
{"region":"AA","env":"","uuid":"foobar","ts":65,"sign":"SIGN","config":[{"key":"httpUrl","need_ca":false},{"key":"mqttUrl","need_ca":false},{"key":"httpsPSKUrl","need_ca":false},{"key":"mqttsPSKUrl","need_ca":false},{"key":"ghostUrl","need_ca":true}]}
[14:15:53.397677 LOG (Client)] ==== End request body ===
[14:15:53.397755 LOG (Server)] Response:  {"caArr":null,"httpUrl":{"addr":"http://10.42.42.1/d.json","ips":["10.42.42.1"]},"mqttUrl":{"addr":"10.42.42.1:1883","ips":["10.42.42.1"]},"ttl":600}

Jun  1 14:20:15 dnsmasq-dhcp[15]: 220324384 available DHCP range: 10.42.42.10 -- 10.42.42.40
[...]
```

Ellipsis mine. From here on it repeats the DHCP cycle but not the request-game.

corbolais, 2023-06-01:
> The profile you are using appears correct, and changing profile won't make any difference. CloudCutter is responding correctly when it gets requests, it simply looks like the device isn't making the requests it should be, which usually means wifi issues. There are no tricks here I can recommend, as the device is already on the correct network, it just stops doing what it should do at some point. Try moving devices around to see if there is any orientation they communicate better in?

Yes, I think the device should continue talking. Whether a reorientation improves stuff.. Dunno, pityful wifi then. Will try.

Thanks for your feedback.

Cossid, 2023-06-11:
@corbolais Check to see if any other network interface on your machine is using a 10.42. prefix.  You can usually do this with the `route` command.  Other users have run into issues on machines also running tailscale for example, which appears to use this range.

corbolais, 2023-06-14:
@Cossid Thanks for coming back to me. 

Using `ip r` the output shows no network of the given prefix.

Cossid, 2023-07-09:
Closing as stale.

closed 2023-07-09 by Cossid

corbolais, 2023-07-09:
I'm still working on this though.

Cossid, 2023-07-09:
Sure, but it's not really an open issue for CloudCutter itself, there is nothing on the CloudCutter side that can change.  This appears specific to your device/environment.  After DHCP negotiation, it's completely on the device to make all subsequent requests.

corbolais, 2023-07-09:
That's one way to look at it. 

Since the bulb should be cuttable and ought to be supported it still is of concern to this IMO.

Clearly, something is amiss and others might come to the same conclusion at some point. So this is still unresolved.

Gandalf1783, 2023-08-28:
Idea: Do you have auto-update turned on?
Tuya Devices tend to have a auto update switch. It might never start to fetch if disabled. This might also solve my problem (#463)

Gandalf1783, 2023-08-28:
Also, the fix of #386 might be helpful

corbolais, 2023-08-29:
Thx, will look into it using those hints of yours.