tuya-cloudcutter/tuya-cloudcutter #413: FEIT Electric WIFI Smart Dimmer Switch - DUMP

engineerAndyy, 2023-06-10:
Attached is the dump for the [FEIT Electric WIFI Smart Dimmer Switch]. Can sometimes be found at COSTCO. (https://www.amazon.com/Feit-Electric-Wi-Fi-Smart-Dimmer/dp/B09VJYVXL5/ref=sr_1_6?crid=20SIIGQYMD27W&keywords=feit&qid=1686429772&sprefix=fei%2Caps%2C263&sr=8-6).

[ltchiptool_bk72xx_2023-06-10_12-12-29_Feit_Dimmer_SW.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11713326/ltchiptool_bk72xx_2023-06-10_12-12-29_Feit_Dimmer_SW.zip)

Cossid, 2023-06-10:
This is a dump of LibreTiny/ESPHome, not the original firmware.

engineerAndyy, 2023-06-10:
I'm confused, what's the process for getting the original firmware dump? I used the ITCHIPTOOL v4.1.1 and selected Read flash.

Cossid, 2023-06-10:
You dumped the firmware after you had already flashed ESPHome to the device.  An original dump can only be done before flashing.

engineerAndyy, 2023-06-10:
Shoot.. I must have gotten my boards mixed up during the process.

engineerAndyy, 2023-06-12:
Hello, I have a new dump that I just took from a dimmer switch fresh from Amazon. Let me know if this one works. Thanks. 
[ltchiptool_bk72xx_2023-06-11_12-12-29_Feit_Dimmer_SW.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11728680/ltchiptool_bk72xx_2023-06-11_12-12-29_Feit_Dimmer_SW.zip)

Cossid, 2023-06-12:
Added as `Feit DIM-WIFI Dimmer Switch`.  This is a TuyaMCU device, so there is no GPIO device configuration data.

closed 2023-06-12 by Cossid

dasb00ter, 2024-11-07:
> Hello, I have a new dump that I just took from a dimmer switch fresh from Amazon. Let me know if this one works. Thanks. [...]

@engineerAndyy were these devices flashable using cloudcutter or did you have to go the serial route? After dismantling I can see this is the Bk7231N version from costco.
Also very importantly to me did you have access to control of the light within the switch itself (for finding it in the dark) with any 3rd party software?