tuya-cloudcutter/tuya-cloudcutter #414: Requesting new code for merkury led strip lights MI-EW011-999WW

Packers87, 2023-06-11:
I'm not sure if I am requesting this correctly but, I have the MI-EW011-999WW and i would like to use cloud cutter as I am trying to go fully localized for my light control. I don't know how to build yaml code. I do see that there is code for the other Merkury led strips but not the "advanced" model. I would like to ask for help in getting the code created so i can copy and paste it into my ESP home docker to run cloud cutter, and free it from its jail cell. 

Thanks.

Cossid, 2023-06-11:
That device hasn't been dumped yet, and without a dump, there isn't much that CloudCutter can do.  It looks like that device might be RGBIC instead of RGBW, but I'm not certain.  If that is the case, it may not be fully supported by 3rd party firmware yet.  There is also no guarantee any existing profiles will work with it.  If you were able to find a profile and flash it, you'll need to look into esphome-kickstart and ltchiptool to see if you're able to pull the storage and see how Tuya configured it.

Closing this ticket, as ESPHome config is outside the scope of CloudCutter.  If you are able to provide an original firmware dump for this device, please feel free to attach it and re-open this issue.

closed 2023-06-11 by Cossid

mchipser, 2024-04-26:
I have this device and a dump if you need it..

Cossid, 2024-04-26:
Feel free to post it, and if it's not patched I'll add a CloudCutter profile.  Either way, the device being RGBIC won't currently be fully compatible with ESPHome, but might work with OpenBeken.

mchipser, 2024-04-26:
That is what I have on it now..

Cossid, 2024-04-28:
Added as `Merkury Innovations MI-EW001-999WW RGBIC Strip v1.2.1`

Packers87, 2024-05-03:
> Added as Merkury Innovations MI-EW011-999WW RGBIC Strip v1.2.1

Thanks, I had completely forgotten about this. I'll look into what has to
happen next in the near future. it's been a while since I've looked at the
cloud cutter road.

Appreciate the dump and the info.

Cossid, 2026-04-15:
> > Added as `Merkury Innovations MI-EW011-999WW RGBIC Strip v1.2.1`
> 
> update to 1.2.4 offered https://images.tuyaeu.com/smart/firmware/upgrade/bay1597022670130iZEz/1639194387e6fbf15c724.bin
> 
> ```
> < TUYA IOT SDK V:1.0.2 BS:40.00_PT:2.2_LAN:3.3_CAD:1.0.2_CD:1.0.0 >
> < BUILD AT:2020_09_25_17_24_52 BY embed FOR ty_iot_wf_bt_sdk_bk AT bk7231t >
> ```

Added as `Merkury Innovations MI-EW011-999WW RGBIC Strip v1.2.4`