tuya-cloudcutter/tuya-cloudcutter #416: SC400W curtain switch flash dump

vxav, 2023-06-13:
I couldn't find a brand for this but it has a WB3S chip.
P/N: b8ttSC400W-V5T

- [Link to product page](https://fr.aliexp
[ltchiptool_bk72xx_2023-06-13_17-47-04.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11736839/ltchiptool_bk72xx_2023-06-13_17-47-04.zip)
ress.com/item/1005003223090512.html?spm=a2g0o.order_list.order_list_main.41.119f5e5bkCbEdJ&gatewayAdapt=glo2fra)

Cossid, 2023-06-13:
Added as `Tuya Generic SC400W Curtain Switch`

closed 2023-06-13 by Cossid

vxav, 2024-10-02:
Well that sucks, I ordered that device again (1+ year later) as one died and I noticed it is now a CB3S chip so that wb3s  profile doesn't work.

I dumped the flash with `ltchiptool` and when trying to build the profile it fails. Could I be doing something wrong?

[Tuya-Generic_SC400W-CB3S-Curtain-Switch.bin.zip](https://github.com/user-attachments/files/17230755/Tuya-Generic_SC400W-CB3S-Curtain-Switch.bin.zip)

```
> python3 build_profile.py Tuya-Generic_SC400W-CB3S-Curtain-Switch.bin
[+] Processing file='Tuya-Generic_SC400W-CB3S-Curtain-Switch.bin' as Tuya-Generic_SC400W-CB3S-Curtain-Switch
RBL containers:
	0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xea20]
		extracted to /Users/xavier/gh/tuya-cloudcutter/profile-building/Tuya-Generic_SC400W-CB3S-Curtain-Switch
	0x129f0a: app - [encoding_algorithm=NONE, size=0xdba80]
		extracted to /Users/xavier/gh/tuya-cloudcutter/profile-building/Tuya-Generic_SC400W-CB3S-Curtain-Switch
Storage partition:
	0x1ee000: 28 KiB - 6 keys
	- 'gw_bi'
	- 'gw_di'
	- 'tyDevDataOffset'
	- 'wf_start_md'
	- 'gw_wsm'
	- 'tyDevDataIndex'
		extracted all keys to /Users/xavier/gh/tuya-cloudcutter/profile-building/Tuya-Generic_SC400W-CB3S-Curtain-Switch/Tuya-Generic_SC400W-CB3S-Curtain-Switch_storage.json
App code `user_param_key`:
	- not found!
[+] Searching for known exploit patterns
[!] The binary supplied appears to be patched and no longer vulnerable to the tuya-cloudcutter exploit.
```

I guess that means the only way is to serial flash it with e.g. ltchiptool?

Cossid, 2024-10-02:
> Well that sucks, I ordered that device again (1+ year later) as one died and I noticed it is now a CB3S chip so that wb3s profile doesn't work.
>
> I dumped the flash with `ltchiptool` and when trying [...]

Correct.

vxav, 2024-10-03:
I had to toy around to find the right pins as there are differences between the config of the BK7231T and N as I wasn't sure how to get the current config (including pin assignment) from the dumped bin but it was pretty straightforward in the end.