tuya-cloudcutter/tuya-cloudcutter #417: Deta 6000HA with WB2S flash dump

clydebarrow, 2023-06-20:
I have a Deta 6000HA inline switch (manufactured by Arlec) that uses a WB2S module. I have been able to dump the flash from the module, but when running `build_profile` I can't get a token - it stalls at `[+] Waiting for multicast token from app...` and appears to be listening on UDP port 6669. Wireshark (on a different computer, same network) shows no traffic on that port when I follow the instructions in the Smart Life app (V4.8.2.) There are broadcasts from the Smart Life app on port 7000 but these start as soon as the app opens so don't seem to be related. I have tried a 

I've attached the flash dump if someone else can create a profile from it. Or is there another way to get a token?

```
[+] Processing file='Deta_6000HA-inline-switch.bin' as Deta_6000HA-inline-switch
[+] Encrypted bin has already been extracted
[+] Haxomatic has already been run
[+] uuid: 2bd98e9fbc4d115e
[+] auth_key: bArj5V7TnZ2ccQr9mJIOY2qRUD3FZ0qU
[+] ap_ssid: GRID
[+] swv: 1.0.1
[+] bv: 30.05
[+] firmware key: keyvp8kp9h9ew9nt
[+] product key: keyvp8kp9h9ew9nt
[!] No gw_di, No version or key stored, manual lookup required
[+] SDK: 2.0.0
[+] Device class: oem_bk7231s_1_plug_arlec
[!] No token provided.
[!] On any device on the same network as you're device running this script, please log into the Smart Life app ('Try as Guest' works fine if you do not already have an account)
[!]  Note: no real device should currently be in pairing mode, or the instructions below may not match.
[!]  - Start the add device procedure (hit '+' in the upper-right and select 'Add Device')
[!]  - Under 'Add Manually' select 'Socket (Wi-Fi)'
[!]  - Enter your network credentials as instructed
[!]  - Do not follow any of the instructions about putting a device into pairing mode, instead select 'next' until it asks the status of the indicator and select 'Blink Slowly'
[!]  - Select 'Go to Connect', then in your wifi selection screen, hit the back button to return to Smart Life.
[!] A new token should be sent to your network, and this script will continue.
[!] Note: this will join an unresponsive device to your account.  You can safely delete it afterwards.
[+] Waiting for multicast token from app...
```

[Deta_6000HA-inline-switch.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11796514/Deta_6000HA-inline-switch.bin.zip)

Cossid, 2023-06-20:
Added as `DETA 6000HA Inline Switch`.  Unfortunately, this is a device-specific firmware and there is no pin assignment information present, so you'll need to manually scan for pin assignments on your own.

closed 2023-06-20 by Cossid

clydebarrow, 2023-06-21:
I confirmed that the pin assignments matching those from the older version works. Here's my ESPHome config for the GPIO - not sure how to translate this into the cloudcutter device profile.

```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO11
      inverted: True
      mode: INPUT_PULLUP
    name: ${name}
    filters:
      delayed_on_off: 100ms
    on_press:
      then:
        - switch.toggle: relay

switch:
  - platform: gpio
    pin: GPIO26
    name: ${name}
    id: relay
    restore_mode: RESTORE_DEFAULT_OFF
    icon: mdi:power-socket-au
    on_turn_on:
      then:
        - output.turn_on: button_led
    on_turn_off:
      then:
        - output.turn_off: button_led

output:
  - platform: gpio
    id: button_led
    pin:
      number: GPIO6
      inverted: yes
```

Disperion, 2024-12-11:
Hi
You were able to get a token tuya?