tuya-cloudcutter/tuya-cloudcutter #418: Lost API password/key - Reflash possiblity

robertscheib, 2023-06-20:
Long story short, I was able to successfully cut a flash ESP on a Mercury Innovations light and connect it to HA. Subsequently I lost the API password/key. I have full access to the light and it's still connected to HA but I cannot flash an update YAML.

Do I have any recovery options or am I stuck with it's current configuration at this point?

closed 2023-06-20 by Cossid

Cossid, 2023-06-20:
> Can't you recover the API password from HA?

API key and OTA password are generally different.  You can probably recover the API key from your Home Assistant's `.storage` directory `esphome.<randomid>` file, but it won't have the OTA password to force another update.

Cossid, 2023-06-21:
> I was hoping there would be a way to use it to flash the "original" kickstart firmware again and or move to OpenBeken (and then back?). If that isn't possible with CloudCutter, maybe I can use the OTA. 

No, once you flash a 3rd party firmware, there is no longer anything CloudCutter can do.  CloudCutter only works with devices that have stock Tuya firmware.

> I do have the OTA password and access, but I'm unsure how to convert the `.bin` file to `ug` or `uf2`.

If you still have the OTA password, what exactly is your problem?  The uf2 file is generated along side the .bin, you can grab it from `.esphome/build/<devicename>/.pioenvs/<devicename>/firmware.uf2

robertscheib, 2023-06-21:
Wouldn't that version be what is currently on the device, including the lost API password. I was going to try and flash the firmware that Cloudcutter puts on first. Am I looking at things incorrectly here?

Cossid, 2023-06-21:
CloudCutter doesn't put anything on 'by default', you have to choose what it puts on, but once it's flashed 3rd party firmware on a device once, CloudCutter can't do anything to that device again.  Your only remaining options are whatever OTA method your 3rd party firmware supports, or serial flashing.

If you haven't recompiled ESPHome, yes, that uf2 file would be what is already on the device, but if you set a new API key (keeping the OTA password the same) and recompile, it would be different, and you should be able to re-add the device to HA with the new api key.