tuya-cloudcutter/tuya-cloudcutter #428: Antela fs1302-UK

deltamelter, 2023-07-02:
Somebody else claims to have flashed this device with BK7231N image but was not able to dump anything or access the power monitoring options.  I want to flash several of these and would like to avoid opening.
I don't have my lightleak esp with me so cannot try that until next week so was wondering if I can confirm the chipset with cloudcutter??

If I tuya cut with wrong chip type and firmware version will it behave differently and would I be able to re-cut with alternative chip type?

The Tuya app shows the following versions, which are not listed.  Any recommendations for how to proceed?
[image]

Cossid, 2023-07-02:
Unfortunately, Plug version 1.1.12 is on the [known patched list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) and serial flashing is the only option.

closed 2023-07-02 by Cossid

CharlieTemplar, 2023-07-02:
is

> Unfortunately, Plug version 1.1.12 is on the [known patched list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) and serial flashing is the only option.

Oh, I wondered about it not being in the cutting list.  I that definitely patched version in all cases?
Would it do any harm to try?  Would trying cloudcutter stop the tuya app from working if the exploit doesn't work?

Cossid, 2023-07-02:
All plugs seen with h 1.1.12 and 1.1.15 have been patched.  No, it won't hurt anything to try, but it is nearly certainly not vulnerable.

deltamelter, 2023-07-03:
I have 2 boxes of these plugs.  1 box are marked 202204HS and get further with tuya-cloudcutter with 1.1.8 profile but now stuck
the other box are all marked 202301HS, and it is one of those that gives 1.1.12

202204HS - don't get past this
[image]

202301HS
[image]

Cossid, 2023-07-03:
The 1.1.8 are vulnerable, and indeed, the exploit works, but something network/timing related is preventing it from finishing.  Leave cloudcutter running, put the device into pairing mode, and open smart life and attempt a manual add, and instead of connecting it to your network, connect it to the ap `cloudcutterflash` with the password `abcdabcd`.  It will not connect/add to smart life, but cloudcutter should pick back up where it left off and finish.

The 1.1.12 are almost certainly flashed, and as I stated, serial flashing will be the only option.

deltamelter, 2023-07-03:
Didn't continue, now I have 2 of these that are "half-way"  they now AP as A-XXXX which cloudcutter seems to accept if I try again but are no longer usable with the the tuya-app :(

Connecting to cloudcutter on the phone with the tuya-app shows:
[image]

But the plug doesn't seem to register...
Any ideas on what the "network/timing" issue might be?

Cossid, 2023-07-03:
They won't connect to the smart life app, but the instructions listed above using the smart life app should aid in allowing cloudcutter to finish.  Make sure your target device and cloudcutter devices are near each other, as AP mode is not very strong.

deltamelter, 2023-07-03:
Maybe because I connected to the cloudcutterflash AP first, other times I connected to it from the app, but seems it completed
but not visible on the wifi SSID that I specified on cmdline
[image]
Will this routine be the same if I flash instead of just cut?

Cossid, 2023-07-03:
If you flash, it depends on the functionality of what you flash, but you can only flash once with cloudcutter.  The same procedure will help you flash it it got stuck in the same spot.  You can cut as many times as you like to try different wifi settings, until you flash.

deltamelter, 2023-07-03:
is something cached after the device is cut?
Can't re-cut now
[image]

deltamelter, 2023-07-04:
I still have a copyof the deviceConfig file with the random keys used when this was successful.  how do I use that to connect and re-cut?
I wonder if I mis-Typed the target wifi because the device doesn't show up where I told it

Cossid, 2023-07-04:
You don't need any previous information to recut, you simply just re-run the process.  You must select the same profile that originally worked, or you will receive errors, but that has nothing to do with the saved configs.

deltamelter, 2023-07-04:
I think I may have tried a different  profile AFTER receiving errors.
How can I recover the 2 plugs from this?  Both in A-XXXX AP ssid "state" and one was difficult to get complete the subsequent cut always hanging on the last part and eventually got to complete but with key error above.
I tried to recut the other but cannot get it past that last part, no matter what I try with cloudcutterflash and the tuya app.
So do I need to clear something or reset somehow?

Cossid, 2023-07-04:
No, there is nothing you can reset.  The procedure is exactly the same whether cutting or re-cutting, but after it's been cut, we can't verify if you're using the correct profile or not, because it already has the `A-` ssid prefix.

Just choose by firmware version and name, select 1.1.8, then select any device presented, it doesn't matter.  Make sure your devices are nearby to the device running cloudcutter, and use the app to re-send cloudcutterflash ap instructions if it hangs in the same place.  There is nothing that can change in that process.

deltamelter, 2023-07-04:
Thanks, I will give it a try. I only chose different 1.1.8 options, just different device profiles.
I can't get the device and AP cloudcutterflash much closer :)
[image]

The device that does respond eventually doesn't seem to do so in direct reaction to the tuya app, but eventually hits iot.dns.tuya.com.

deltamelter, 2023-07-04:
[image]

deltamelter, 2023-07-04:
only ever chosen between the LSPA7 and LSPA9 devices and when this was cut it was with LSPA9
but still get this:
[image]

deltamelter, 2023-07-04:
I still have the `/configured-devices/ ` deviceConfig with the random localkey used when the cut was successful, this is no longer the most recent file in there because I tried to re-cut so many times with the 2 plugs so far.
Is there a way to use this key and details? or is something else now written to the device? (does the target wifi ssid and this key only get written to the device after that successful cut, or every time since?)  I an now fairly sure I typed the SSID last 2 digits incorrect which is probably why it didn't show on my router?  How do I fix this?

Cossid, 2023-07-05:
No, if the device never connected to your network, those keys and details are useless.  Cutting again will just change them.

Are you sure you have your devices correct and do not have two of them plugged in at the same time?  The only way you can get PSKKEY/AUTHKEY do not match is if you cut with the correct profile, and later tried cutting with an incorrect profile, or the wrong device is connecting in the second phase.

deltamelter, 2023-07-05:
Only one device at a time, but I have been switching between 2 of them.  I looked back through the deviceConfig files, and at some point I have tried a different firmware version number.  
So it there any way out of these 2 states?
1 device won't go to that last piece, can't get it to complete past the last cloudcutterflash connection.  (Anything I can do to trying to force that with tinytuya or other tool?)
1 device tells me incorrect keys..
I have 2 more untouched devices of the same version, but don't want to risk "bricking" those too

Cossid, 2023-07-05:
I've already guided you through all the steps/solutions that exist.  The errors you show for PSKKEY/AUTHKEY can only result from you doing something incorrect.  The device that won't connect to the last stage, you just have to keep trying to send the cloudcutterap info via smart life until it works.

deltamelter, 2023-07-05:
OK, so I did something incorrect, there is no way out of that other than serial flash?  The keys have moved on out of expected sequence and now as good as lost.
with the other device I tried several times.  Do I just keep putting it in fast blink and going thru the motions of choosing the plug, choosing the wifi?  Do you know which step in this sends the "iot" POST which leads to completion?

Cossid, 2023-07-05:
You just need to start over.  If you're selecting the correct profile, you should never get PSKKEY/AUTHKEY errors, because it's writing new ones each time and using those, but if you use the wrong profile, it's not properly overwriting them, but continuing on because it was successfully exploited at some point in the past.

Slow blink is usually the best way, especially if you have multiple devices, though fast blink should work as well for when it is stuck unable to connect to `cloudcutterap`

deltamelter, 2023-07-05:
The device that wasn't completing I tried a few times last night but gave up and just left cloudcutterflash running and the device plugged in.  Today I totally forgot about it until I just checked the tmux console and after 14 hours after I started it just decided to complete and the device shows up on my network. can I find anything/ try it out with tinytuya or should I just go for the flash? 

Ok, with the second device that gives the incorrect key, how do you mean start over? I first cut with 1.1.8 profile and may have tried other versions since successful cut.  I have only tried 1.1.8 device profiles since then  and fails every time.  Am I misunderstanding what you mean by profile?

Thanks for your helpful responses so far on this.

Cossid, 2023-07-06:
> The device that wasn't completing I tried a few times last night but gave up and just left cloudcutterflash running and the device plugged in. Today I totally forgot about it until I just checked [...]

The choice is yours, you can try tinytuya with the keys it gave you at the end of the process, or you can flash, in which case the keys don't matter (it will assign new random ones when you start the flash process).

> Ok, with the second device that gives the incorrect key, how do you mean start over? I first cut with 1.1.8 profile and may have tried other versions since successful cut. I have only tried 1.1.8 device profiles since then and fails every time. Am I misunderstanding what you mean by profile?

You need to post a full console log, preferably with the `-v` option, because there is no way you should get a SECKEY/AUTHKEY error if you're selecting the correct profile, and the screenshots you have posted so far don't provide enough data to verify anything.

deltamelter, 2023-07-06:
[cloudcutter-2023-07-06_0226.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11963458/cloudcutter-2023-07-06_0226.txt)
Used:
`./tuya-cloudcutter.sh -v  -s <SSID> <wifiPSK>`

Since tried with
`./tuya-cloudcutter.sh -v -p tuya-generic-lspa9-plug-v1.1.8 -s <SSID> <wifiPSK>`

Cossid, 2023-07-06:
Stop connecting to `cloudcutterflash` with your phone, it will not help in any way, it just makes the logs a huge mess.

Are you certain the 1.1.8 devices are also BK7231N?  There is also a BK7231T 1.1.8 plug firmware (rnd_switch).  What I'm seeing in the log is that it is using the same key it tried to set, but it probably failed, but it doesn't recognize it because the device already has an `A-` AP prefix.  Do an ls on your `device-profiles` directory to see what profiles you have tried so far, and post here so they can be compared.

Can you zip and attach `configured-devices/SU6aC2SN7JnS.deviceconfig` so we can verify it is setting valid values in there?  Further, could you zip and attach your entire `configured-devices` directory so we can see the progression of what was actually attempted?

deltamelter, 2023-07-06:
> Stop connecting to `cloudcutterflash` with your phone, it will not help in any way, it just makes the logs a huge mess.

Only connecting the phone after 5-10 mins of waiting for something else to happen.  You suggested using the tuya-app to trigger the last part, I only have tuya-app on the phone.  I'll try without the phone.   (It has been running for 45 mins so far with no sign of connecting to `cloudcutterflash`, should I at least but the plug in fast blink mode, or just leave it?)

> Are you certain the 1.1.8 devices are also BK7231N? There is also a BK7231T 1.1.8 plug firmware (rnd_switch). What I'm seeing in the log is that it is using the same key it tried to set, but it [...]

I got 2 boxes of plugs, one box is newer and patched, the other box are all the same older version.  This one device with the wrong key was the same one I cut successfully (https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/428#issuecomment-1618717473) I used the same BK7231N 1.1.8 device profile to cut it, but failed since.
It's an identical plug from the same box, so I have no reason to suspect it to be BK7231T.
I did backup both the `device-profiles` directory and the `configured-devices` folder at one point to see if it helped to start from scratch in case something had changed in those that could be affecting things

`device-profiles1:
total 44
-rw-r--r--  1 pi   pi   1147 May 22 13:02 README.md
-rw-r--r--  1 pi   pi     22 May 22 13:02 .gitignore
drwxr-xr-x  2 pi   pi   4096 May 22 13:02 schema
drwxr-xr-x  2 root root 4096 May 25 02:25 tuya-generic-bsd31-16a-smart-plug
drwxr-xr-x  2 root root 4096 Jul  3 00:16 tuya-generic-lspa7-plug
drwxr-xr-x  2 root root 4096 Jul  3 00:19 tuya-generic-lspa9-plug-v1.1.8
drwxr-xr-x  2 root root 4096 Jul  3 00:25 tuya-generic-lspa9-plug-v1.0.3
drwxr-xr-x  9 pi   pi   4096 Jul  3 00:36 .
drwxr-xr-x  2 root root 4096 Jul  3 00:36 feit-bpplug-smart-3-smart-plug
drwxr-xr-x 14 pi   pi   4096 Jul  6 03:25 ..

device-profiles:
total 16
drwxr-xr-x  2 root root 4096 Jul  4 01:31 tuya-generic-lspa9-plug-v1.1.8
drwxr-xr-x  4 root root 4096 Jul  4 16:56 .
drwxr-xr-x  2 root root 4096 Jul  4 16:56 tuya-generic-lspa7-plug
drwxr-xr-x 14 pi   pi   4096 Jul  6 03:25 ..`
 
The different devices you can see are what I tried when the cut would not complete past cloudcutterflash and I was under the impression that cutting was harmless. I don't remember trying ` "1.0.3 - BK7231T / oem_bk7231s_dltj"` or why?! but it's there, in black and white, but was before actual cut and after initial cut attempt.  Eventually after advice in this thread I successfully cut using `tuya-generic-lspa9-plug-v1.1.8` but fairly sure now that I transposed the last 2 digits of the wifi ssid on the commandline

> 
> Can you zip and attach `configured-devices/SU6aC2SN7JnS.deviceconfig` so we can verify it is setting valid values in there? Further, could you zip and attach your entire `configured-devices` directory so we can see the progression of what was actually attempted?

[SU6aC2SN7JnS.deviceconfig.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11967440/SU6aC2SN7JnS.deviceconfig.zip)
[configured-devices.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11967622/configured-devices.zip)

deltamelter, 2023-07-07:
> Sorry, I forgot the SmartLife app forces you to connect in order to send (and this isn't normally a part of the process). I guess you have to connect. Slow-blink is the preferred method still.

Slow blink mode in the app doesn't seem to trigger the final part, but fast blink does... eventually....

> Based on what I've seen so far, I'm not certain your plugs are actually 1.1.8, but there is no way to verify now that they have been exploited. What the full log you posted tells me is you tried the LSPA9-v1.1.8 profile, it tried setting data via the exploit, and when it tried to use that data, it failed, which means that attempted exploit was not a correct match.

I still have 2 from the same box, same markings that are so far untouched. I will verify version with tuya-app.  I also have access to lightleak esp now in case it is useful to try the N profiles from there.  Attached all the logs for the attempts with all of the other profiles. an they are all the same :(
I also have one from this box, now successfully cut with 1.1.8 and happily accessible with tinytuya

> It is harmless when you don't mix up device profiles and provide correct data. The problem here is you appear to be getting mixed up. The thing is, the device doesn't change until an exploit is successfully run, but afterwards, if you re-exploit, there is no way to verify the profile you are currently using is correct, because it always looks successful because of the `A-` prefix from the previous exploit.

You mention that _"the device doesn't change until an exploit is successfully run"_ so does that mean that the device would connect to an AP that have the same SSID that I mistyped?  Is it possible to get the device to connect to that?  I wonder if it would be easy to use the same method as tuya-cloudcutter uses and monitor is the device connects to an AP with the last 2 digits transposed?  Also I was not at home all week so have been doing this on a different network, so might need a similar method to access the cut plug  to change the wifi ssid for the one here...

> I'm still questioning this. Based on the logs you have provided, I'm not certain LSPA9 v1.1.8 is what actually exploited the device. None of the screenshots you provided confirm this, they were all too cut off to verify. At this point, I would certainly not recommend flashing until you are absolutely sure which profile actually works for the device (though it would likely fail with the same PSKKEY/AUTHKEY failure, but if we're not certain this is T or N at this point, it wouldn't be safe). I would recommend trying to cut again using the the `tuya-generic-lspa9-plug-v1.0.3` (BK7231T v1.0.3) and `feit-bpplug-smart-3-smart-plug` (BK7231N v1.1.5) profiles since they are present around the same timeframe, it can't be ruled out they are valid. Provide full logs of each run if you are able.

I can't imagine that this one device is T and the another in the same box is proven N type, but I suppose it could be the case.
Anyway, here are the logs for the attempts with the other profiles that I had used... 
 
[cloudcutter-feit-bpplug-smart-3-smart-plug-2023-07-06_1607.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11973331/cloudcutter-feit-bpplug-smart-3-smart-plug-2023-07-06_1607.txt)
[cloudcutter-tuya-generic-lspa7-plug-2023-07-06_1137.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11973332/cloudcutter-tuya-generic-lspa7-plug-2023-07-06_1137.txt)
[cloudcutter-tuya-generic-lspa9-plug-v1.0.3-2023-07-06_1622.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11973333/cloudcutter-tuya-generic-lspa9-plug-v1.0.3-2023-07-06_1622.txt)

As an aside I have also opened one of the newer patched plugs and the good news is that they are not glued or sonic welded, just held in with 4 tabs.  The bad news it that the cb2s card is mounted vertically and the pins are not "easily" accessible.  I might have to try to build a tool to reach the pins...  I don't want to destroy or cut a hole in these
[image]
[image]
[image]
[image]

Cossid, 2023-07-07:
> Slow blink mode in the app doesn't seem to trigger the final part, but fast blink does... eventually....

That is strange, they should mostly both work the same way.

> You mention that _"the device doesn't change until an exploit is successfully run"_ so does that mean that the device would connect to an AP that have the same SSID that I mistyped? Is it possible to get the device to connect to that?

Maybe, it depends on what is actually happening.  If the exploit is failing each time, the device is still in the last saved state, and if  you take it out of pairing mode, it might try the last settings that were saved.  However, if the exploit is still working, but working incorrectly, we can't know for sure.

> I can't imagine that this one device is T and the another in the same box is proven N type, but I suppose it could be the case. Anyway, here are the logs for the attempts with the other profiles [...]

Yeah, it's unlikely it's a T device, as all those logs fail in the exact same way, I just wanted to try and be able to rule that out.

> As an aside I have also opened one of the newer patched plugs and the good news is that they are not glued or sonic welded, just held in with 4 tabs. The bad news it that the cb2s card is mounted [...]

That's somewhat fortunate, but yeah, the CB2S is kind of a pain all-around.  I'm guessing the board is attached to that bottom white plastic piece via the power prongs and not just clipped in.  Since they are somewhat easy to open (while not being easy to flash), it would be a good idea to open the problematic devices and ensure they are indeed CB2S as well.  As for attempting to flash, building a tool is going to be very difficult, as all the pins you need are on the even-less-accessible back side of the board.

Lastly, for cutting, it is possible to specify a device id and local key with the `-d` and `-l` paramters, perhaps try re-using the values that were supplied when you first successfully exploited the device.  See `./tuya-cloudcutter.sh -h` for more info.  I'm running out of ideas for what is happening here, this (failing re-exploiting an exploited device) isn't something that has been seen before.

deltamelter, 2023-07-08:
> That's somewhat fortunate, but yeah, the CB2S is kind of a pain all-around. I'm guessing the board is attached to that bottom white plastic piece via the power prongs and not just clipped in. Since they are somewhat easy to open (while not being easy to flash), it would be a good idea to open the problematic devices and ensure they are indeed CB2S as well (Edit: or is it only the 2023 model that opens like that?). As for attempting to flash, building a tool is going to be very difficult, as all the pins you need are on the even-less-accessible back side of the board.
> 
> Lastly, for cutting, it is possible to specify a device id and local key with the `-d` and `-l` paramters, perhaps try re-using the values that were supplied when you first successfully exploited the device. See `./tuya-cloudcutter.sh -h` for more info. I'm running out of ideas for what is happening here, this (failing re-exploiting an exploited device) isn't something that has been seen before.

The opening is "easier" than other devices, but still tricky to do without damage.
I just opened the device giving me the issues above and part of the "tab" next to the tuya board snapped off, it it just an internal piece of plastic, so  not a problem, but it allowed me to see that there is actually quite of space around the pins and it I cut away a wider strip of the thin white plastic, I will be able to get me long bent pin tool around the back of the board to the 3v3, tx and rx and flash it.

Nothing else has worked so for this device and the 4 patched devices it's serial all the way...

deltamelter, 2023-07-08:
btw, what is the method to change the wifi ssid and password for a working cut device, or do I risk recutting it?
Followed this but I didn't follow. 
https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/346

Cossid, 2023-07-08:
> btw, what is the method to change the wifi ssid and password for a working cut device, or do I risk recutting it? Followed this but I didn't follow. #346

It has always been just re-cutting, but you can also do it with tinytuya, in fact Tuya-CloudCutter does just that as the last step when cutting: https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/bdecb710bad1c058b2885f63b021b980a6bf5a69/src/cloudcutter/__main__.py#L39

deltamelter, 2023-07-08:
> That's somewhat fortunate, but yeah, the CB2S is kind of a pain all-around. I'm guessing the board is attached to that bottom white plastic piece via the power prongs and not just clipped in. Since [...]

I meant to post pics and confirmation that (so far as labels and pcb print say) All versions are identical hardware wise and CB2S.  I managed to use my tool on the above problem device.  The backup confirms it was 1.1.8 and the wifi ssid was "cloudcutterflash" so I think that means that the wifi ssid write at the end of the cut didn't work.

[image]
[image]

Cossid, 2023-07-08:
If you have the dump, feel free to post it, might give clues to why re-cutting might not have worked?

deltamelter, 2023-07-09:
[readResult_BK7231N_QIO_2023-08-7--18-41-57.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11993512/readResult_BK7231N_QIO_2023-08-7--18-41-57.zip)

Cossid, 2023-07-09:
I'm not sure what to tell you at this point.  It had a different auth key than any you posted logs for, so it was likely changing it all along.  I flashed it to a board and cut it, and it cut the first time without error, and after I rebooted it, it joined right up to my network.

deltamelter, 2023-07-09:
> I'm not sure what to tell you at this point. It had a different auth key than any you posted logs for, so it was likely changing it all along. I flashed it to a board and cut it, and it cut the first time without error, and after I rebooted it, it joined right up to my network.

I made a few more attempts to cut on different network and trying different things since I gave you the directory dump
So something specific to the hardware or the tuya-cloudcutter image/env  This is from may and was used to more smoothly cut a T device back then. Had  installed other stuff on that pi since and may have "broken" something for tuya-cloudcutter install?  
Might try fresh install.  

The next 2 I will try to flash directly using cloudcutter exploit and then I have this one that I cut successfully, currently on a different wifi ssid so I will reluctantly try to flash that one with cloudcutter too. just worried that it will go the same way if I re-cut and I want to avoid opening any more than needed....

deltamelter, 2023-07-11:
Just don't know what it is with these devices....  spent 2hours not connecting to cloudcutterflash and the flash seems to be continung, but damn slow...

`[18:36:51.442975 Firmware Upload] /files/OpenBK7231N_UG_1.17.180.bin send complete, request range: bytes=0-632655/632656
[19:00:44.018569 Log (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.timer.count&devId=j6hToxQyWaXxeOcqjWJV&et=1&t=1689102043&v=4.0&sign=ae18aba54bb7026958e7e773ae5742c7', version='HTTP/1.1', remote_ip='10.42.42.15')
[19:00:44.019294 LOG (Client)] ==== Request body ===
{"devId":"j6hToxQyWaXxeOcqjWJV","lastFetchTime":"0","t":1689102043}
[19:00:44.021735 LOG (Client)] ==== End request body ===
[19:00:44.024006 LOG (Server)] Response:  {'result': {'devId': 'j6hToxQyWaXxeOcqjWJV', 'count': 0, 'lastFetchTime': 0}, 't': 1689102044, 'success': True}
[19:25:44.403480 Log (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.timer.count&devId=j6hToxQyWaXxeOcqjWJV&et=1&t=1689103544&v=4.0&sign=98fced9c73c7a166cda61e5c62c6da36', version='HTTP/1.1', remote_ip='10.42.42.15')
[19:25:44.404059 LOG (Client)] ==== Request body ===
{"devId":"j6hToxQyWaXxeOcqjWJV","lastFetchTime":"0","t":1689103544}
[19:25:44.404231 LOG (Client)] ==== End request body ===
[19:25:44.406974 LOG (Server)] Response:  {'result': {'devId': 'j6hToxQyWaXxeOcqjWJV', 'count': 0, 'lastFetchTime': 0}, 't': 1689103544, 'success': True}
[19:29:34.874994 MQTT Received] Topic: log/runstat/j6hToxQyWaXxeOcqjWJV - Message: {"time":1689103774,"sum-t":1689107374,"mem":45904,"rssi":-53}
[19:30:27.741543 MQTT Received] Topic: smart/device/out/j6hToxQyWaXxeOcqjWJV - Message: {"protocol":4,"t":1689103827,"data":{"dps":{"20":2381}}}
`

I this actually still working?
Con you tell from these scarce messages?

Cossid, 2023-07-11:
There really seems to be something wrong with your network if things are taking more than about 3-5 minutes.

> [18:36:51.442975 Firmware Upload] /files/OpenBK7231N_UG_1.17.180.bin send complete, request range: bytes=0-632655/632656

Indicates it indeed sent the entire file, but it clearly never rebooted if it's doing timer pings 25 minutes later.  Try rebooting the device and see if it comes back up with TuyaOS or OpenBeken installed.

deltamelter, 2023-07-11:
What is supposed to happen after the firmware upload? 
The timer pings are from the device or the cloudcutter script?

Cossid, 2023-07-11:
Timer pings are from the device, cloudcutter is only receiving them and responding that it received them.

There should have been MQTT status messages stating progress of flashing the firmware, but it appears those never came (partial logs don't help verify, but the fact it never exited means it never got them).  After that, it just exits and waits for the firmware to apply and reboot, which usually takes up to a couple minutes (though some devices apparently don't reboot on their own and need to physically be rebooted, but they still should have sent the MQTT status if it processed the firmware upgrade).  At this point, cloudcutter isn't doing anything other than responding to messages it gets from the device and sending the firmware data (as it did) when specifically requested.  There is no world in which it should take more than 5 minutes to complete a firmware upgrade after making a dhcp request to `cloudcutterflash`, and if it does, it's because of the network conditions or hardware problems.

deltamelter, 2023-07-11:
after restarting the device it stopped communicating with the cloudcutterAP.   Then I lost ssh to the pi running the cloudcutter so lost the shell, tho it is still running.
Put the device in AP mode and it shows as A-XXXX, so I guess still tuya-OS. 
Should this be still in a cloudcuttable condition to try again, or is it likely gone too far?
It must be the device hardware.  I was hoping to get get more than one of these flashed without opening, but I guess I'm going to have to better at opening and serial flashing these... but probably going to have one more to practice on now... :(

Cossid, 2023-07-11:
You can try cutting/flashing again.  As long as it boots back to TuyaOS, you can always try again.
You can't flash a partial image, it has built-in verification checks when it applies the updates (the bootloader does).