tuya-cloudcutter/tuya-cloudcutter #431: AP Connection fails

clydebarrow, 2023-07-09:
I'm trying to cut a Deta 6000HA and get to where cloudcutter is trying to connect to WiFi, and get this failure repeated:

```
Found access point name: "GRID-F615", trying to connect...
Error: Connection activation failed: (11) 802.1X supplicant took too long to authenticate.
....
```

This is running on Ubuntu on RPi.

I can successfully connect to the AP as a regular client from a different machine.

Any ideas?

Cossid, 2023-07-09:
Make sure the target device and device running CloudCutter are near by each other.  AP mode is from either device is not very strong.  That is the OS unable to connect, not CloudCutter, there is nothing that can be changed in the way it connects.

deepcoder, 2023-07-09:
Park in your backpack, "reboot after every pass no matter how far you get down the path: success, failure or you just punt at some point".

clydebarrow, 2023-07-09:
They are right next to each other and I have tried rebooting with no success.

deepcoder, 2023-07-09:
There is a fine art to this process. I have a raspberry pi 4 dedicated to the Cloud cutter and tuya cloud apps, sitting right at my lab desk. On my desktop machine I use an app to watch the 2.4 Ghz wifi for Ap's that appear and disappear. I have a large 'stash' of cheap wine and beer that I drink while using a small A/C power switch to toggle the power to the target device. Patience, booze and a trigger finger that has been the right 'equation'.

And I still have a drawer of 'failures' :-(

Good hunting!

clydebarrow, 2023-07-09:
Thanks for the input. I think I'll just fall back to serial flashing. Sadly these units require the WB2S to be desoldered since the serial pins are in use for other things. But that's going to be quicker than endlessly retrying the cutter approach.

The main reason for using these units is that they have substantial screw terminals on them. I've also used Shelly in-line switches which are easy enough to use in native mode or re-flash, but they have pissy little screw terminals that are IMHO quite unsuited to mains wiring.

closed 2023-07-09 by clydebarrow