tuya-cloudcutter/tuya-cloudcutter #432: bricked devices

tbgoose, 2023-07-09:
Hi all

I have a few Arlec PC191HA plugs running firmware 1.35 that I have tried to flash ESPHome onto.

The process went smoothly and everything looked like it worked, but now the devices do not turn on / respond at all upon being plugged in. I am guessing that whilst tuya-cloudcutter seemed to succeed somewhere along the line the firmware got scrambled and the devices are now bricked.

Any pointers on if they are recoverable? Have tried power cycling them lots of times and different variations of holding down the button whilst plugging them in etc.

Cossid, 2023-07-09:
Depends one how the firmware you flashed was configured.  If it was for the wrong chip, serial is the only recovery method (though current CloudCutter has pretty strong checks in place to disallow this from happening).  If you didn't have captive_portal or ap enabled, you'll also probably need serial recovery.

If you had captive portal or ap mode enabled, they should auto-recover to that state with some time.  Without logs or console output, there isn't much further advice we can give.  This is another device that has both a T and an N version, but if you have a current checkout of CloudCutter, it should have been enforcing the correct firmware with the correct profile.

Do you have the config you compiled?

tbgoose, 2023-07-10:
Yes I used this profile. Pretty sure that was the correct one.

https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/arlec-pc191ha-smart-plug-bk7231t-v1.3.5.json

Cloudcutter was pulled only yesterday, so def running the current version

Cossid, 2023-07-10:
I was asking for your esphome config, not which cloudcutter profile you used.

If you used the included kickstart firmware, it won't connect to your network, it will start up it's own AP mode.

tbgoose, 2023-07-10:
Right, oops. I can see the AP now and have added it to my iot network. Once
I go to its address I can see I can upload a new firmware.

This is the esp-home config I think I want to use

> esphome:
>   name: router-socket
>   friendly_name: router_socket
>
> libretuya:
>   board: generic-bk7231t-qfn32-tuya
>   framework:
>     version: dev
>
> # Enable logging
> logger:
>
> # Enable Home Assistant API
> api:
>   encryption:
>     key: "TQNsIbF5ELVwbEYM9PVi8O/nsh4vxs6XWdUNF6hHFLM="
>
> ota:
>   password: "a20ed677ec825ff6c3147ba9f1440a9c"
>
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>
>   # Enable fallback hotspot in case wifi connection fails
>   ap:
>     ssid: "Router-Socket Fallback Hotspot"
>     password: "czA78vtHh58U"
>
> {
>   "vendor": "Arlec",
>   "bDetailed": "0",
>   "name": "Arlec Grid Connect PC191 Smart Plug-In Socket with Energy
> Meter",
>   "chip": "BK7231T",
>   "board": "WB2S",
>   "pins": {
>     "6": "Rel;1",
>     "7": "BL0937CF;1",
>     "8": "BL0937CF1;1",
>     "10": "WifiLED_n;1",
>     "11": "Btn;1",
>     "24": "BL0937SEL;1",
>     "26": "LED;1"
>   },
>   "keywords": [
>     "switch",
>     "4 pack",
>     "Bunnings Australia",
>     "BL0937",
>     "two relays, cuts off both N and L"
>   ],
>   "image": "https://obrazki.elektroda.pl/2567152300_1672151363.png",
>   "wiki": "https://www.elektroda.com/rtvforum/topic3944452.html",
>   "command": ""
> }
>

From here <https://openbekeniot.github.io/webapp/devicesList.html>. I don't
quite understand how to go from having the yaml to flashing the config. Do
I just go to ip of the plug, click OTA Update and select the
router-socket.yaml I linked above?

tbgoose, 2023-07-10:
Right, oops. I can see the AP now and have added it to my iot network. Once I go to its address I can see I can upload a new firmware.

This is the esp-home config I think I want to use
```
    esphome:
      name: router-socket
      friendly_name: router_socket

    libretuya:
      board: generic-bk7231t-qfn32-tuya
      framework:
        version: dev

    # Enable logging
    logger:

    # Enable Home Assistant API
    api:
      encryption:
        key: "TQNsIbF5ELVwbEYM9PVi8O/nsh4vxs6XWdUNF6hHFLM="

    ota:
      password: "a20ed677ec825ff6c3147ba9f1440a9c"

    wifi:
      ssid: !secret wifi_ssid
      password: !secret wifi_password

      # Enable fallback hotspot in case wifi connection fails
      ap:
        ssid: "Router-Socket Fallback Hotspot"
        password: "czA78vtHh58U"

    {
      "vendor": "Arlec",
      "bDetailed": "0",
      "name": "Arlec Grid Connect PC191 Smart Plug-In Socket with Energy Meter",
      "chip": "BK7231T",
      "board": "WB2S",
      "pins": {
        "6": "Rel;1",
        "7": "BL0937CF;1",
        "8": "BL0937CF1;1",
        "10": "WifiLED_n;1",
        "11": "Btn;1",
        "24": "BL0937SEL;1",
        "26": "LED;1"
      },
      "keywords": [
        "switch",
        "4 pack",
        "Bunnings Australia",
        "BL0937",
        "two relays, cuts off both N and L"
      ],
      "image": "https://obrazki.elektroda.pl/2567152300_1672151363.png",
      "wiki": "https://www.elektroda.com/rtvforum/topic3944452.html",
      "command": ""
    }
```

From [here](https://openbekeniot.github.io/webapp/devicesList.html). I don't quite understand how to go from having the yaml to flashing the config. Do I just go to ip of the plug, click OTA Update and select the router-socket.yaml I linked above?

Cossid, 2023-07-10:
You can use [UPK](https://upk.libretiny.eu/) to select your profile and generate a full ESPHome yaml.  You need to compile the yaml and use the resulting `.uf2` file to upload to OTA update.

closed 2023-07-10 by Cossid