tuya-cloudcutter/tuya-cloudcutter #436: ALD095HA RGBCT Downlight Issues

lordvorta, 2023-07-25:
Hey guys I have been battling trying to get this working on some ALD095HA RGBCT downlights. Initial I thought I was having some issues with my wireless adapter a TP-Link Archer T600U on a Raspberry Pi2. However I have since scraped that setup and tried running cloud cutter with a new Raspberry 4b with its built in Wifi adapter running raspberry pi os.

I am getting the same result across both setups and two different lights that I have tried custom firmware installation. I am able to get the exploit to run initially and can see the device's Wifi network change to AP mode the device successfully connects to the cloudcutter SSID. However the device doesn't get any further it appears that it constantly sits doing DCHP request but never gets any further. I am not sure what is happening any assistance would be greatly appreciated. Logs below of a failed attempt. 

lordvorta@raspberrypi4:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -v
Building cloudcutter docker image
[+] Building 15.7s (13/13) FINISHED                                                                                                            docker:default
 => [internal] load build definition from Dockerfile                                                                                                     0.7s
 => => transferring dockerfile: 460B                                                                                                                     0.0s
 => [internal] load .dockerignore                                                                                                                        1.4s
 => => transferring context: 2B                                                                                                                          0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-buster                                                                             5.0s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:830e161433edfe047a23ebc99c12ee0eb1dc0a50e6b5f1c98e869ac271786632                  0.0s
 => [internal] load build context                                                                                                                        0.1s
 => => transferring context: 3.02kB                                                                                                                      0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iprou  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                            0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                      0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                 0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                             0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                               0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                0.0s
 => exporting to image                                                                                                                                   0.1s
 => => exporting layers                                                                                                                                  0.0s
 => => writing image sha256:6afbb0251e7da05d02b3dcb284d635a7f990239a264dabb27b1c06f7d48d4d86                                                             0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                           0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Arlec
   Alantop
   Aldi
   Alic
   Amico
   Antela
   Aoycocr
 > Arlec
   Atarm
   Athom
   Atomi
   Aubess
   Avatar
   Avatto

[?] Select the article number of your device: ALD095HA RGBCT Downlight
   ALD092CHA CCT Downlight
 > ALD095HA RGBCT Downlight
   GLD12x HA Series 2 RGB E27 Bulb
   GLD130HA
   Grid Connect GLD132HA Bulb
   PC191HA Smart Plug BK7231N v1.1.8
   PC191HA Smart Plug BK7231T v1.3.5
   RGB B22 Bulb

[?] Select the firmware version and name: 2.9.16 - BK7231T / oem_bk7231s_light_ty
 > 2.9.16 - BK7231T / oem_bk7231s_light_ty

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
[... 107 lines cut ...]
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  1 option: 53 message-type  5
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 51 lease-time  12h
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 58 T1  6h
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 59 T2  10h30m
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option:  1 netmask  255.255.255.0
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option:  3 router  10.42.42.1
Jul 25 10:25:30 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option:  6 dns-server  10.42.42.1
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 client provides name: wlan0
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 DHCPREQUEST(wlan0) 10.42.42.36 d8:1f:12:23:6d:92 
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 tags: wlan0
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 DHCPACK(wlan0) 10.42.42.36 d8:1f:12:23:6d:92 wlan0
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 next server: 10.42.42.1
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  1 option: 53 message-type  5
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 51 lease-time  12h
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 58 T1  6h
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 59 T2  10h30m
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option:  1 netmask  255.255.255.0
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option:  3 router  10.42.42.1
Jul 25 10:25:34 dnsmasq-dhcp[15]: 1116131344 sent size:  4 option:  6 dns-server  10.42.42.1

Cossid, 2023-07-25:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

lordvorta, 2023-07-25:
Perfect! that seemed to have done the trick put the light in ap mode again and manually joined it via the smartlife app to the cludcutter ssid and the custom firmware flashed successfully.

closed 2023-07-25 by lordvorta