tuya-cloudcutter/tuya-cloudcutter #437: I need some help with BK7231N CBLC5 hardware

matijacob96, 2023-07-26:
I've bought several RGB+CW GU10 smart bulbs and i'm having several problems. I tried to run the tuya-cloudcutter but it gets stuck on:

```
root@raspberrypi:/home/user/tuya-cloudcutter# sudo ./tuya-cloudcutter.sh -s IoT-Sweethome 4LF4J0r3s
Building cloudcutter docker image
[+] Building 1.1s (13/13) FINISHED                                                                       docker:default
 => [internal] load build definition from Dockerfile                                                               0.0s
 => => transferring dockerfile: 460B                                                                               0.0s
 => [internal] load .dockerignore                                                                                  0.0s
 => => transferring context: 2B                                                                                    0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-buster                                       0.8s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:830e161433edfe047a23ebc99c12ee0eb1dc0a50e6  0.0s
 => [internal] load build context                                                                                  0.0s
 => => transferring context: 3.02kB                                                                                0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dn  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                      0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                           0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                       0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                         0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                          0.0s
 => exporting to image                                                                                             0.0s
 => => exporting layers                                                                                            0.0s
 => => writing image sha256:f4050cfd24e709f2dbfcfc6bf96d633de52d58692ae357e6e546e878c5659b93                       0.0s
 => => naming to docker.io/library/cloudcutter                                                                     0.0s
Successfully built docker image
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.3.21 - BK7231N / oem_bk7231n_light_ty
   1.3.0 - BK7231N / kltwfb_single_winctrl_rf_lns00
   1.3.1 - BK7231N / oem_bk7231n_light_ty
   1.3.19 - BK7231N / oem_bk7231n_light_ty
   1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.3.2 - BK7231T / oem_bk7231s_dltj
   1.3.2 - BK7231T / oem_bk7231s_light_pir_globe
 > 1.3.21 - BK7231N / oem_bk7231n_light_ty
   1.3.5 - BK7231T / oem_bk7231s_dltj
   1.6.0 - BK7231T / oem_bk7231s_curtain_switch
   2.0.0 - BK7231N / oem_bk7231n_irbox_mol_ty
   2.0.0 - BK7231N / oem_bk7231n_smart_ir_rf_pro_ty
   2.0.0 - BK7231N / oem_bk7231n_strip_ty
   2.0.0 - BK7231T / oem_bk7231s_light5_sm2135e_yamao

[?] Select the brand of your device: Tuya Generic
   BNETA IoT
   Doogain
   Feconn
   Firefly
   Globe Electric
   Immax
   Kanlux
   LEDLUMEN
   Laser
   Lumary
   Mirabella Genio
   Smarteck
 > Tuya Generic

[?] Select the article number of your device: SWWFD1C412 RGBCT Downlight
   1947Z5R W509Z1 RGBCT Bulb
 > SWWFD1C412 RGBCT Downlight

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: tuya-generic-swwfd1c412-rgbct-downlight
Selected Profile: oem-bk7231n-light-ty-1.3.21-sdk-2.3.3-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
........
Found access point name: "A-A2CA", trying to connect...
Device 'wlan0' successfully activated with 'e8d39b3c-6325-450d-897a-2a4bccbc9592'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/vmXNb8mFPrDE.deviceconfig
Saved device config in /work/configured-devices/vmXNb8mFPrDE.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "A-A2CA", trying to connect...
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
..
Found access point name: "A-A2CA", trying to connect...
Device 'wlan0' successfully activated with 'e8d39b3c-6325-450d-897a-2a4bccbc9592'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Cutting device off from cloud...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr d8:3a:dd:0d:01:01 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
```

I have no idea how to continue from here and I don't know if it is because the model selected.

Cossid, 2023-07-26:
Start here: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

matijacob96, 2023-07-26:
> Start here: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do
> 
> Also make sure the machine running CloudCutter does not already have a 10.42.x.x routing pattern/ip range.

```
SSL Error on 15 ('10.42.42.33', 58978): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('10.42.42.33', 58986): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('10.42.42.33', 58988): [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1129)
404 GET /generate_204 (10.42.42.33) 2.40ms
SSL Error on 15 ('10.42.42.33', 58108): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 17 ('10.42.42.33', 58114): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET / (10.42.42.33) 2.63ms
404 GET / (10.42.42.33) 5.55ms
SSL Error on 18 ('10.42.42.33', 58128): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 18 ('10.42.42.33', 58134): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 19 ('10.42.42.33', 58138): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET / (10.42.42.33) 2.12ms
```

Cossid, 2023-07-26:
Those are your phone connecting to the cloudcutter AP, any time you see `GET /` you can ignore all requests that include the IP that followed it.  That is just part of the process, since the Smart Life app likes to connect before it sends the credentials along to the device.  Let it continue and see if the target device eventually connects.  You may consider running with `-v` as it will provide extra debug information, especially around the DHCP stage, about devices attempting to connect.

matijacob96, 2023-07-26:
> Those are your phone connecting to the cloudcutter AP, any time you see `GET /` you can ignore all requests that include the IP that followed it. That is just part of the process, since the Smart [...]

```
Using WLAN adapter: wlan0
Jul 26 21:29:52 dnsmasq[15]: started, version 2.80 cachesize 150
Jul 26 21:29:52 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jul 26 21:29:52 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Jul 26 21:29:52 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Jul 26 21:29:52 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr d8:3a:dd:0d:01:01 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
[21:29:55.280487 MQTT] Connected
Jul 26 21:30:01 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:30:01 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPDISCOVER(wlan0) cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPOFFER(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  2
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPDISCOVER(wlan0) cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPOFFER(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  2
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPREQUEST(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPACK(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  5
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
```

matijacob96, 2023-07-26:
```
wlan0: AP-ENABLED
[21:29:55.280487 MQTT] Connected
Jul 26 21:30:01 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:30:01 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPDISCOVER(wlan0) cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPOFFER(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  2
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPDISCOVER(wlan0) cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPOFFER(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  2
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPREQUEST(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 tags: wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPACK(wlan0) 10.42.42.13 cc:8c:bf:e5:a2:ca wlan0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  5
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Jul 26 21:32:06 dnsmasq-dhcp[15]: 225386433 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:32:06 dnsmasq-dhcp[15]: 225386433 vendor class: android-dhcp-13
Jul 26 21:32:06 dnsmasq-dhcp[15]: 225386433 client provides name: S23-Ultra-de-Mati
Jul 26 21:32:07 dnsmasq-dhcp[15]: 225386433 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:32:07 dnsmasq-dhcp[15]: 225386433 vendor class: android-dhcp-13
Jul 26 21:32:07 dnsmasq-dhcp[15]: 225386433 client provides name: S23-Ultra-de-Mati
Jul 26 21:32:09 dnsmasq-dhcp[15]: 225386433 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:32:09 dnsmasq-dhcp[15]: 225386433 vendor class: android-dhcp-13
Jul 26 21:32:09 dnsmasq-dhcp[15]: 225386433 client provides name: S23-Ultra-de-Mati
Jul 26 21:32:11 dnsmasq-dhcp[15]: 1879886770 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jul 26 21:32:11 dnsmasq-dhcp[15]: 1879886770 vendor class: android-dhcp-13
Jul 26 21:32:11 dnsmasq-dhcp[15]: 1879886770 client provides name: S23-Ultra-de-Mati
Jul 26 21:32:14 dnsmasq-dhcp[15]: 1879886770 DHCPDISCOVER(wlan0) ea:29:a2:3d:88:f7
Jul 26 21:32:14 dnsmasq-dhcp[15]: 1879886770 tags: wlan0
Jul 26 21:32:14 dnsmasq-dhcp[15]: 1879886770 DHCPOFFER(wlan0) 10.42.42.33 ea:29:a2:3d:88:f7
[... cut ...]
Jul 26 21:34:40 dnsmasq[15]: config connectivitycheck.gstatic.com is 10.42.42.1
Jul 26 21:34:40 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:40 dnsmasq[15]: config www.google.com is 10.42.42.1
Jul 26 21:34:40 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:40 dnsmasq[15]: config www.google.com is 10.42.42.1
SSL Error on 15 ('10.42.42.33', 44676): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
```

And it stills repeating this until bulb stops blinking.

Cossid, 2023-07-26:
> Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPDISCOVER(wlan0) cc:8c:bf:e5:a2:ca
> Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 tags: wlan0
> Jul 26 21:30:04 dnsmasq-dhcp[15]: 842012285 DHCPOFFER(wlan0) [...]

This shows the device tried to connect and made 3 DHCP requests, but never progressed from the ACK phase.  All you can do is make sure the devices are as nearby each other as possible and keep retrying.

matijacob96, 2023-07-26:
> Jul 26 21:34:25 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
> Jul 26 21:34:25 dnsmasq[15]: config www.google.com is 10.42.42.1
> Jul 26 21:34:25 dnsmasq[15]: query[A] [...]

```
Jul 26 21:34:25 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:25 dnsmasq[15]: config www.google.com is 10.42.42.1
Jul 26 21:34:25 dnsmasq[15]: query[A] connectivitycheck.gstatic.com from 10.42.42.33
Jul 26 21:34:25 dnsmasq[15]: config connectivitycheck.gstatic.com is 10.42.42.1
Jul 26 21:34:25 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:25 dnsmasq[15]: config www.google.com is 10.42.42.1
SSL Error on 15 ('10.42.42.33', 42698): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
Jul 26 21:34:27 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:27 dnsmasq[15]: config www.google.com is 10.42.42.1
Jul 26 21:34:27 dnsmasq[15]: query[A] connectivitycheck.gstatic.com from 10.42.42.33
Jul 26 21:34:27 dnsmasq[15]: config connectivitycheck.gstatic.com is 10.42.42.1
Jul 26 21:34:27 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:27 dnsmasq[15]: config www.google.com is 10.42.42.1
SSL Error on 15 ('10.42.42.33', 51998): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
Jul 26 21:34:32 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:32 dnsmasq[15]: config www.google.com is 10.42.42.1
Jul 26 21:34:32 dnsmasq[15]: query[A] connectivitycheck.gstatic.com from 10.42.42.33
Jul 26 21:34:32 dnsmasq[15]: config connectivitycheck.gstatic.com is 10.42.42.1
Jul 26 21:34:32 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:32 dnsmasq[15]: config www.google.com is 10.42.42.1
SSL Error on 15 ('10.42.42.33', 52006): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
Jul 26 21:34:40 dnsmasq[15]: query[A] connectivitycheck.gstatic.com from 10.42.42.33
Jul 26 21:34:40 dnsmasq[15]: config connectivitycheck.gstatic.com is 10.42.42.1
Jul 26 21:34:40 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:40 dnsmasq[15]: config www.google.com is 10.42.42.1
Jul 26 21:34:40 dnsmasq[15]: query[A] www.google.com from 10.42.42.33
Jul 26 21:34:40 dnsmasq[15]: config www.google.com is 10.42.42.1
SSL Error on 15 ('10.42.42.33', 44676): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
```

Have you seen the next comment?

Cossid, 2023-07-26:
That's all your phone, it's irrelevant.

matijacob96, 2023-07-26:
```
[21:57:48.897079 Log (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.timer.count&devId=CXk84JxhGWaTR4AGBhEu&et=1&t=1690408667&v=4.2&sign=6a07e13fb770aa4a5f4e4eef1fd9667b', version='HTTP/1.1', remote_ip='10.42.42.13')
[21:57:48.897283 LOG (Client)] ==== Request body ===
{"devId":"CXk84JxhGWaTR4AGBhEu","lastFetchTime":"0","t":1690408667}
[21:57:48.897423 LOG (Client)] ==== End request body ===
[21:57:48.899292 LOG (Server)] Response:  {'result': {'devId': 'CXk84JxhGWaTR4AGBhEu', 'count': 0, 'lastFetchTime': 0}, 't': 1690408668, 'success': True}
[21:57:48.928024 MQTT Received] Topic: smart/device/out/CXk84JxhGWaTR4AGBhEu - Message: {"protocol":16,"data":{"progress":"94","firmwareType":0}}
[21:57:49.436342 MQTT Received] Topic: smart/device/out/CXk84JxhGWaTR4AGBhEu - Message: {"protocol":16,"data":{"progress":"98","firmwareType":0}}
Firmware file has been sent and MQTT reported a progress of nearly complete.  Waiting 15 seconds to ensure flashing completes.
Flashing should be complete.  It takes about 15 seconds for the device to reboot and verify the flash was valid.
Please wait about 30 seconds then look for signs of activity from the firmware you supplied (either watch for AP mode or check if it joined your network).
Device MAC address: cc:8c:bf:e5:a2:ca
```

Where could I find my device now?

Cossid, 2023-07-26:
Depends on what firmware you flashed.  Either way, it should be broadcasting its own AP for you to connect to and join it to your network.

matijacob96, 2023-07-27:
Everything has worked correctly. Thanks

closed 2023-07-27 by matijacob96