tuya-cloudcutter/tuya-cloudcutter #442: Smart WiFi Switch 2 Gang dump - ATS02L-US (WB3S)  - Need profile build

Jackie-P, 2023-08-02:
Hi everyone,

I finally succeeded to read the flash of a  Qualitel ATS03L 2 Gang light switch with "Lightleak".  
It worked for me like described in the documentation using the LightLeak BK7231T profile. Great work, thank you

These are TUYA lightswitches built in South Africa, but is basically the same as one I bought from AliExpress as well. Before the chips were upgraded it had TYWE2S chips which could easily be flashed with Tasmota or Esphome

Now I need help to build the profile from dump that is attached.

Can somebody please help and create the profile for me? Once done I will try and flash these light switches (about 12 of them using CloudCutter and load ESPHOME onto them

Thank you very much 

Jackie

CPU: [WB3S](https://developer.tuya.com/en/docs/iot/wb3s-module-datasheet?id=K9dx20n6hz5n4)
Dump: [dump_2gang-light.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12243602/dump_2gang-light.zip)
Image: [image]

Cossid, 2023-08-02:
Added as `Tuya Generic ATS02L-US 2 Gang Touch Switch`

closed 2023-08-02 by Cossid

Jackie-P, 2023-08-03:
Thank you very much. 
Everything worked 100%, and now 4 of my 2 Gang devices had been loaded with ESPHOME (LibreTiny)