tuya-cloudcutter/tuya-cloudcutter #444: Smart WiFi Switch 1 Gang dump - ATS01L-US (WB3S) - Need profile build

Jackie-P, 2023-08-03:
Hi everyone,

I had successfully read the flash of a Qualitel ATS01L 1-Gang light switch with "Lightleak".
It worked for me like described in the documentation using the LightLeak BK7231T profile. Great work, thank you

These are TUYA lightswitches built in South Africa, but is basically the same as one I bought from AliExpress as well. Before the chips were upgraded it had TYWE2S chips which could easily be flashed with Tasmota or Esphome

Now I need help to build the profile from dump that is attached.  Can somebody please help and create the profile for me? 

Thank you very much

Jackie

CPU: [WB3S](https://developer.tuya.com/en/docs/iot/wb3s-module-datasheet?id=K9dx20n6hz5n4)
Dump: [dump_1gang-light.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12249096/dump_1gang-light.zip)

Info: https://templates.blakadder.com/qualitel_ATS01L-US.html

Cossid, 2023-08-03:
Added as `Tuya Generic ATS01L-US 1 Gang Touch Switch`

closed 2023-08-03 by Cossid

Jackie-P, 2023-08-04:
Thank you very much! Working perfectly!