tuya-cloudcutter/tuya-cloudcutter #446: Doogain Q5-E12 RGBCW Bulb 1.3.21 not flashing

ehonzay, 2023-08-13:
I cannot seem to flash this device, looks like MQTT has a problem, but could not find anything useful. I've tried 3 diffferent wifi adaptors across two systems, error is always the same. Same error also occurs with another bulb from same batch. 

sudo ./tuya-cloudcutter.sh -v -r
Building cloudcutter docker image
Sending build context to Docker daemon  28.54MB
Step 1/10 : FROM python:3.9.12-slim-buster AS base
 ---> ead817e27369
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> af2b317c3806
Step 3/10 : FROM base AS python-deps
 ---> af2b317c3806
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> da479b623146
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 7defdd12fa96
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 92ff4ba932c6
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 86c30ae306fa
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 86c30ae306fa
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 1109887c0a96
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 8eafe845c3bc
Successfully built 8eafe845c3bc
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Doogain
   Crabtek
   Cree
   DETA
   DOGAIN
   Daybetter
   Dewenwils
 > Doogain
   ECASA
   ELEGRP
   EarthBulb
   Earu
   EasyLiv
   Enbrighten

[?] Select the article number of your device: Q5-E12 RGBCW Bulb
 > Q5-E12 RGBCW Bulb

[?] Select the firmware version and name: 1.3.21 - BK7231N / oem_bk7231n_light_ty
 > 1.3.21 - BK7231N / oem_bk7231n_light_ty

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 4980.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.07.[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.17.130_bk7231n.ug.bine-Kickstart-v23.07.13_bk7231n_app.ota.ug.bin
   ESPHome-Kickstart-v23.07.13_bk7231n_app.ota.ug.bin
 > OpenBeken-v1.17.130_bk7231n.ug.bin
   OpenBK7231N_UG_1.17.212.bin

Selected Device Slug: dogain-q5-e12-rgbcw-bulb
Selected Profile: oem-bk7231n-light-ty-1.3.21-sdk-2.3.3-40.00
Selected Firmware: OpenBeken-v1.17.130_bk7231n.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
....
Found access point name: "A-E2F6", trying to connect...
Device 'wlp2s0' successfully activated with '177fc548-6078-4277-91ed-3f3f706156e1'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/Uii2atKhnd06.deviceconfig
Saved device config in /work/configured-devices/Uii2atKhnd06.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
...........................
Found access point name: "A-E2F6", trying to connect...
Device 'wlp2s0' successfully activated with '04848b2c-e597-4e97-85ea-20dd3ca6409e'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlp2s0
Aug 13 15:24:25 dnsmasq[15]: started, version 2.80 cachesize 150
Aug 13 15:24:25 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Aug 13 15:24:25 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Aug 13 15:24:25 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlp2s0
Aug 13 15:24:25 dnsmasq[15]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
Using interface wlp2s0 with hwaddr b8:8a:60:23:19:50 and ssid "cloudcutterflash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 401, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 243, in __update_firmware
    __configure_local_device_or_update_firmware(args, update_firmware=True)
  File "/src/cloudcutter/__main__.py", line 106, in __configure_local_device_or_update_firmware
    mqtt.mqtt_connect(device_id, local_key, tornado.ioloop.IOLoop.current(), graceful_exit_timeout=flash_timeout, verbose_output=args.verbose_output)
  File "/src/cloudcutter/protocol/mqtt.py", line 68, in mqtt_connect
    client.connect(broker)
  File "/src/.venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/src/.venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/src/.venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.9/socket.py", line 844, in create_connection
    raise err
  File "/usr/local/lib/python3.9/socket.py", line 832, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
Oh no, something went wrong with updating firmware! Try again I guess...

Cossid, 2023-08-13:
You are correct, it appears that mqtt is failing, so it is unrelated to the target device and your wifi card.  CloudCutter runs both a MQTT server and MQTT client.  The log shows port 1883 is available, but this log also shows that the MQTT server either is not starting, or if it is, firewall rules are not allowing anything to connect to it.  Refused seems to indicate it is not starting or binding the port.  Since it starts within a docker container, there is pretty much no way it can't bind to its local 1883 port, so that means docker isn't allowing it to bind to port 1883 on the host.

I don't have any concrete suggestions for you here, as this is a systems level issue.  You'll need to find out why docker is having trouble port binding.

ehonzay, 2023-08-13:
Thanks, I also tried wiping the docker container and letting it rebuild but that yielded the same results. I'll attempt to troubleshoot the docker container.

Cossid, 2023-08-15:
The issue wouldn't be within the docker container, it would be in the network host mapping of the container and that port not truely being available to bind.  MQTT (theoretically) can't fail to start within the container, but it can fail to bind through to the host outside of docker.

ehonzay, 2023-08-15:
Thanks for that, I will look in that direction. Ubuntu 22.04 is what I'm using if anyone has any additional tips, else I'll be googling.

ehonzay, 2023-08-16:
Looks like i had two docker packages installed, not sure how. Once I removed all traces of any docker/container, and re-installed, things are working, and I was able to flash replacement firmware on the bulb.

closed 2023-08-16 by ehonzay