tuya-cloudcutter/tuya-cloudcutter #447: Wifi adaptor: Device not found

SilverHazeLazers, 2023-08-15:
Hi,

I am trying to use cloudcutter and having some trouble configuring it to use my Alfa AWUS036NH wifi adaptor.

When I run `lsusb` from inside the LXC I can see the adaptor with the name `Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter`

However when I try and start cloudcutter with:
`sudo ./tuya-cloudcutter.sh -w 'Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter'`

I receive `Error: Device 'Ralink' not found.`

I suspect it might be due to unescaped characters in the device name however I havnt been able to solve it

Any help would be much appreciated

Cossid, 2023-08-15:
That is the named description of your wifi adapter, not its mapped allocation.  Run `ifconfig` or `ip a` to see the mapped allocation, it will be near the beginning of each section, and usually contains an abbreviated and indexed descriptor like `wlan0`

SilverHazeLazers, 2023-08-15:
Thanks very much @Cossid 

`nmcli --get-values GENERAL.DEVICE,GENERAL.TYPE device show` worked a treat

closed 2023-08-15 by SilverHazeLazers