tuya-cloudcutter/tuya-cloudcutter #449: Trying to flash topgreener - keeps failing - logs attached

badmadscientist, 2023-08-17:
Hello,
My laptop is ubuntu 22.04.2 LTS with 32GB ram.

I have run tuya-cloudcutter.sh many times on two different devices.  When the unit goes into AP mode, the SSID is now A-**** whereas before it was SmartLife-xxxx  - Also, at this point I cannot get them back in SmartLife app, which I understand should be expected, and while I do not believe these devices are bricked, I purchased these extra units specifically to cut them from the tuya cloud, in one way or another as I have around 50 of these around the house.

But aside from that, it appears to fail for some reason; I believe I followed the instructions fully.  And while it was running I opened another terminal and checked for docker:

CONTAINER ID   IMAGE         COMMAND                  CREATED         STATUS         PORTS     NAMES
a4f1b12649de   cloudcutter   "pipenv run python3 …"   3 minutes ago   Up 3 minutes             lucid_perlman

This is the command I am running:

sudo ./tuya-cloudcutter.sh -vs <SSID> <password>

Verbose output:

Building cloudcutter docker image
Sending build context to Docker daemon  27.91MB

Step 1/10 : FROM python:3.9.12-slim-buster AS base
 ---> ead817e27369
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> c927ccd5f04f
Step 3/10 : FROM base AS python-deps
 ---> c927ccd5f04f
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 1bab151277ef
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 9ce4b50f2ada
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 83a3108b7da5
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 8fe049d7ad27
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 8fe049d7ad27
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 3e485df77f4c
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 6790d1b0034e
Successfully built 6790d1b0034e
Successfully tagged cloudcutter:latest
Successfully built docker image
Loading options, please wait...

How do you want to choose the device?: By manufacturer/device name
Select the brand of your device: Topgreener
Select the article number of your device: TGWF115PQM Energy Monitoring Plug
Select the firmware version and name: 1.1.8 - BK7231N / oem_bk7231n_plug

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 7607.
Port 53 is needed to resolve DNS queries

Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: topgreener-tgwf115pqm-energy-monitoring-plug
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
........
Found access point name: "A-3BAD", trying to connect...
Device 'wlp2s0' successfully activated with '2a7d6ad9-9249-4c15-828a-03fa3717b16e'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/IPWSPRkmbKH2.deviceconfig
Saved device config in /work/configured-devices/IPWSPRkmbKH2.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
..
Found access point name: "A-3BAD", trying to connect...
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Wiping NetworkManager configs
.......
Found access point name: "A-3BAD", trying to connect...

Device 'wlp2s0' successfully activated with '1b3ee4e5-bd6a-47ab-8ea1-80265fccdbdb'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Cutting device off from cloud...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
================================================================================

Using WLAN adapter: wlp2s0
Aug 17 22:40:38 dnsmasq[15]: started, version 2.80 cachesize 150
Aug 17 22:40:38 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Aug 17 22:40:38 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Aug 17 22:40:38 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlp2s0
Aug 17 22:40:38 dnsmasq[15]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
Using interface wlp2s0 with hwaddr 20:4e:f6:c3:74:73 and ssid "cloudcutterflash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 401, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 106, in __configure_local_device_or_update_firmware
    mqtt.mqtt_connect(device_id, local_key, tornado.ioloop.IOLoop.current(), graceful_exit_timeout=flash_timeout, verbose_output=args.verbose_output)
  File "/src/cloudcutter/protocol/mqtt.py", line 68, in mqtt_connect
    client.connect(broker)
  File "/src/.venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/src/.venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/src/.venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.9/socket.py", line 844, in create_connection
    raise err
  File "/usr/local/lib/python3.9/socket.py", line 832, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
Oh no, something went wrong with detaching from the cloud! Try again I guess...

Cossid, 2023-08-17:
Your container can't bind the MQTT port (1883) back to the host, which means you may have multiple containers running?  Or something still binding to port 1883.

badmadscientist, 2023-08-18:
Thanks for the response.

I considered the same based on some googling; but I dont appear to have anything on that port:

sudo lsof -i:1883
(nothing returned)

check that the container is running:

docker container ps
CONTAINER ID   IMAGE         COMMAND                  CREATED         STATUS         PORTS     NAMES
b27e407812c9   cloudcutter   "pipenv run python3 …"   4 minutes ago   Up 4 minutes             nostalgic_haibt

note that the docker container doesn't appear to have taken the port, either:

docker container ls --format "table {{.ID}}\t{{.Names}}\t{{.Ports}}" -a
CONTAINER ID   NAMES                     PORTS
b27e407812c9   nostalgic_haibt           

I am not too familiar with docker or what might be wrong.... any more advice on what to look for, @Cossid  ?

Cossid, 2023-08-18:
The docker should not be running all the time, the script does a `docker run` on demand.  If one is stuck running, you should stop it.

badmadscientist, 2023-08-18:
I should have clarified - the docker is only running when the script is running; when the script is ended (failed), then the docker stops.

Cossid, 2023-08-18:
Then something on the host must be binding to 1883.  We have a script that checks, but it's made for RaspberryPI OS, and may not be compatible with all other OSs.  Make sure no MQTT services are running on your host machine.

badmadscientist, 2023-08-18:
I'm at a loss; I may not know much about docker, but I do know there's no MQTT running on my laptop, and at all times, port 1883 is available, and nothing is bound to it....

Cossid, 2023-08-18:
Not sure what to tell you, that error is 100% an MQTT error of some sort, and the only other similar report (https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/446) was because of multiple docker package installs.  Maybe try removing the docker container and start over.

All I can suggest else-wise is try to find a Raspberry PI to run on.

badmadscientist, 2023-08-18:
I appreciate your comments - I'll do some more troubleshooting in the morning. I have an old pi2 - I'll look to see if that's up to the task.  I'll report back anything I find of assistance, so if anyone else runs into this issue......

Cossid, 2023-08-18:
Just a heads-up, pi2 doesn't have wifi built-in, so you'd need a wifi dongle attached for that.

badmadscientist, 2023-08-18:
The Pi2 cannot be located, so I continue with my (Ubuntu) laptop.

You are, of course, correct; MQTT/mosquitto is either not running or at the best case, cannot bind to the port.

I did remove all docker images and containers, and started the process over.  Again, mosquitto was not responding as it was not bound to any port within the container, so for nothing other than sheer interest, I opened a second terminal and started mosquitto with no parameters, not even sudo.

Went back to the main terminal, rerun the command and this time everything seemingly worked; once rebooted, the topgreener plug did find its way onto the new network.

HOWEVER - I would like to know if doing this process, with mosquitto running with no parameters and at best a default config is going to do what we want it to do.

tinytuya did find the unit as well, once it was connected to my IoT wifi.

But then onto the next stage where I try to flash the firmware - that didn't get far at all using the following command:

sudo ./tuya-cloudcutter.sh -p topgreener-tgwf115pqm-energy-monitoring-plug -f ESPHome-Kickstart-v23.07.13_bk7231n_app.ota.ug.bin

Anyway - for now all I'm asking is your opinion on how I started mosquitto in another terminal with no parameters, and if this is not advisable, do you have any recommendations about running mosquitto locally rather than in a docker container, which for reasons to be found out later does not work.

Thanks!

Cossid, 2023-08-18:
As long as mosquitto is configured to allow anonymous users, and it ends up binding to the internal address, all other defaults should be fine.  I believe the default for `allow_anonymous` is false.

https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/572d2786f3baf7b1ab26c4032dae9d4b19c99f6c/src/setup_apmode.sh#L22 is the config we generate that runs inside the docker, which is just

```
listener 1883 0.0.0.0
allow_anonymous true
```
first line binds to port 1883 on all network endpoints (inside the docker, this would include the local 10.42.42.1), second line allows anonymous connections.

So as long as it binds and allows anonymous, that should be enough.

Cossid, 2023-08-20:
Another user has stated the docker version with their ubuntu install was problematic, and using docker from a snap repo resolved their issue.

badmadscientist, 2023-08-20:
Well, that was a good lead; remove the apt docker, installed the snap docker, and everything seemed to work so much better - with that said, once I finished flashing ESPHome-Kickstart-v23.07.13_bk7231n_app.ota.ug.bin (I previously opened the device and confirmed the chip was a Beken BK7231N), the device never woke up again; left it for ten minutes and it didn't reboot; power cycle changed nothing; only a dim blue light, and the switch does not operate the relay.

I'm surprised I managed to seemingly brick it - but I'm really not too bothered; it was worth a try, and perhaps I'll research on unbricking it, even if that means soldering some wires.... any thoughts on that?

But yes - installing the SNAP docker brought everything to life....

kuba2k2, 2023-08-20:
After flashing Kickstart the device looks dead - but you should see the kickstart Wi-Fi access point.

Cossid, 2023-08-20:
After flashing a device with 3rd party firmware, it won't be on your network anymore, unless you specifically uploaded a firmware that already has that information.  If you install ESPHome-Kickstart or OpenBeken, they will start their own AP you have to connect to and tell them how to join your network.  Either will know how to control the device until given configuration to do so.

closed 2023-08-20 by Cossid

badmadscientist, 2023-08-20:
Perfect.  I dont know how I missed the whole kickstart-* WiFi - but it's all working.

Bottom line here was the docker from APT did not work; but the SNAP version works just fine....

Thanks for all the great support.