tuya-cloudcutter/tuya-cloudcutter #451: Failed to build Docker image

wellerode, 2023-08-19:
Hello,
I tried to use tuya-cloudcutter on an lenovo yoga webbook, running lubuntu 18.04 to cut an lsc smart-plug.
I set up a new user "tuy" added to docker and root and copied the unpacked .zip to /home/tuy/temp
Starting the script ends up at 5/10 as shown below:

> tuy@yoga:~/temp$ 
> sudo ./tuya-cloudcutter.sh -d someid -l somepassword -s ssid anotherpassword
> 
> [!] WARNING: Selected wifi AP support: ja
> AP support is manditory for tuya-cloudcutter to work.  If this is blank or 'no' your adapter doesn't support this feature.
> Press any key to continue, or CTRL+C to exitBuilding cloudcutter docker image
> Sending build context to Docker daemon  3.084MB
> Step 1/10 : FROM python:3.9.12-slim-buster AS base
>  ---> ead817e27369
> Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
>  ---> Using cache
>  ---> c0e36f8ee63d
> Step 3/10 : FROM base AS python-deps
>  ---> c0e36f8ee63d
> Step 4/10 : RUN pip install --upgrade pipenv
>  ---> Using cache
>  ---> e24bd5ddabb3
> Step 5/10 : COPY src/Pipfile /src/
> COPY failed: file not found in build context or excluded by .dockerignore: stat src/Pipfile: file does not exist
> Failed to build Docker image, stopping script

What can I do to fix this ?

Greetings
Axel

Cossid, 2023-08-19:
Did you try just running it again?  Otherwise it doesn't have permission to read the files checked out by cloudcutter.  Step 5 is literally just a file copy.

wellerode, 2023-08-19:
I tried it several times and restarted in between.
I did'nt checked the permissions yet. I will check and change them if needed.

wellerode, 2023-08-20:
I changed the permission of ther folder and the files (chmod 777) an tried it two times after a restart with the same result.
Could something be wrong with the files Pipfile is loading or tries to load from url = "https://pypi.org/simple" ?

Cossid, 2023-08-20:
No, the contents of the file are not relevant, step 5 is literally just copying files, not processing them.

Cossid, 2023-08-20:
It sounds like the docker version included with ubuntu via apt might be too old.  Other ubuntu users are having more success with the snap version of docker.

wellerode, 2023-08-21:
I'll try it on another Computer running Lubuntu 22.04.

closed 2023-09-05 by Cossid