tuya-cloudcutter/tuya-cloudcutter #452: How to install on proxmox ct?

BrunoCunha1983, 2023-08-19:
hello i want to install it on proxmox container

Thx
Bruno Cunha
Portugal

Cossid, 2023-08-19:
That won't work.  While it uses docker, it's not a standalone container, there is a shell script that runs, and runs specific parts within a docker container, but it can't be installed on its own.

closed 2023-08-19 by Cossid

BrunoCunha1983, 2023-08-20:
Can you Help me making a install  ,  making a sort off manual to i do the
install
Please i really want to cut my devices from the tuya cloud service
Tks

Cossid, 2023-08-20:
It's literally just a git clone and running the tuya-cloudcutter.sh script, there is nothing else to it.

```
git clone https://github.com/tuya-cloudcutter/tuya-cloudcutter.git
sudo ./tuya-cloudcutter.sh
```