tuya-cloudcutter/tuya-cloudcutter #455: New device - Wipro Smart Plug 16A

sachinss123, 2023-08-22:
Hi, I'm new to Github and don't know how to create a pull request. Here is the JSON config for a new device that I recently flashed using cloudcutter. It worked flawlessly! Can you please add this as a new device?

Device Name: Wipro 16A Wifi Smart Plug
Link: https://www.amazon.in/Wipro-Monitoring-Appliances-Microwave-Conditioners/dp/B08HN9Q2SZ/ref=sr_1_4?keywords=wipro+smart+plug+16+amp&qid=1692705502&sprefix=wipro+smart+plug+16%2Caps%2C245&sr=8-4

JSON:

```json
{
    "gw_bi": {
        "uuid": "moyA4WhcopCT",
        "psk_key": null,
        "auth_key": "aYcy90CrAQszrPxn",
        "ap_ssid": "A",
        "ap_passwd": null,
        "country_code": null,
        "bt_mac": null,
        "bt_hid": null,
        "prod_test": false
    },
    "user_param_key": {
        "bt1_lv": 0,
        "bt1_pin": 10,
        "bt1_type": 0,
        "ch1_stat": 2,
        "ch_cddpid1": 9,
        "ch_dpid1": 1,
        "ch_num": 1,
        "chip_type": 0,
        "crc": 57,
        "ele_pin": 7,
        "jv": "1.1.5",
        "lose_vol": 85,
        "module": "WB2S",
        "net_trig": 2,
        "netled1_lv": 0,
        "netled1_pin": 6,
        "netn_led": 0,
        "nety_led": 1,
        "over_cur": 16500,
        "over_vol": 275,
        "reset_t": 5,
        "resistor": 1,
        "rl1_lv": 1,
        "rl1_pin": 26,
        "rl1_type": 0,
        "sel_pin_lv": 1,
        "sel_pin_pin": 24,
        "vi_pin": 8,
        "vol_def": 0
    },
    "gw_wsm": {
        "nc_tp": 4,
        "ssid": "Y2xvdWRjdXR0ZXJmbGFzaA==",
        "passwd": "YWJjZGFiY2Q=",
        "md": 0,
        "random": 0,
        "wfb64": 1,
        "stat": 2,
        "token": "AAAAAA",
        "region": "AA",
        "reg_key": null,
        "dns_prio": 2
    },
    "gw_di": {
        "abi": 0,
        "id": "wKns9Hcth2uxVX7mlny7",
        "swv": "1.3.5",
        "bv": "40.00",
        "pv": "2.2",
[... 487 lines cut ...]
        {
            "mode": "ro",
            "property": {
                "min": 0,
                "max": 1000000,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 25,
            "type": "obj"
        },
        {
            "mode": "ro",
            "property": {
                "type": "bitmap",
                "maxlen": 6
            },
            "id": 26,
            "type": "obj"
        }
    ],
    "em_sys_env": "BK7231S_2M"
}
```

Cossid, 2023-08-22:
I'll add it when I return from vacation in 2 weeks.

Cossid, 2023-09-05:
Added as `Wipro DSP1160 16A Smart Plug`

closed 2023-09-05 by Cossid

apoorv569, 2025-05-21:
> Added as `Wipro DSP1160 16A Smart Plug`

The exploit is failing for me. I have this device only firmware version is `v1.1.7`.

Cossid, 2025-05-21:
Another case of the manufacturer selling the same model with different modules.  You'll need to do by firmware version 1.1.7 instead of by mfg/model.

apoorv569, 2025-05-21:
> Another case of the manufacturer selling the same model with different modules. You'll need to do by firmware version 1.1.7 instead of by mfg/model.

I tried that as well, that also fails.

Cossid, 2025-06-01:
No way we can be certain.  It's also possible they are beken and also patched.