tuya-cloudcutter/tuya-cloudcutter #456: How can I help - MOES - WS-US3-RFW-N

badmadscientist, 2023-08-22:
I apologize if this is not the right place to say this:

I have a few of these switches around the house that I'd love to get cut from the cloud with cloudcutter; I have taken one apart, and have many pictures - bottom line though, the wifi is a CB2S, part # NH-SWQ, which in turn appears to be a BK7231N which we know works.

Link to the product (I love them!) [https://www.amazon.com/gp/product/B09H2NRLGR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1](url)

What, if anything, can I do to help get this device working?

Cossid, 2023-08-22:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

Start by seeing if there is a compatible profile already.  If not, it will require a serial dump of the firmware.  If it does, you can use existing tools (cloudcutter to install kickstart, then kickstart + ltchiptool) to get storage information to generate a device yaml.

closed 2023-08-22 by Cossid

badmadscientist, 2023-08-22:
There's a couple of MOES listed but not the three switch and not the same model number. Is it safe to try these and see if it works? Do you have any references on how I can do a firmware dump. I've been wanting to get more involved in this stuff. Now seems to be a good time to start. Thanks.