tuya-cloudcutter/tuya-cloudcutter #457: New device - MS-103 "Temperature and humidity Smart Switch module" DUMP

Drewix, 2023-08-23:
I tried flashing with other similar versions but it didn't work I would be very grateful if you would be willing to add support for this device

Device name: MS-103 "Temperature and humidity Smart Switch module"

Tuya main module and MCU: V1.2.8
Motherboard model: NH-2TH16-V1.1
Chip Tuya: WB3S (BK7231T)

“pins”: {
“1”: “Btn;0”,
“6”: “Rel;1”,
“7”: “1-Wire;0”, #TO DS18D20 OR AM2301A
“8”: “LED_n;1”,
“9”: “LED_n;0”,
“10”: “WifiLED_n;0”,
“24”: “Rel;0”
},

Thread about this sensor:
https://www.elektroda.com/rtvforum/topic3966365.html

DUMP FILE made with OpenBeken:

[OpenBK7231T_UA_1.17.180.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12422523/OpenBK7231T_UA_1.17.180.zip)

kuba2k2, 2023-09-02:
The attached file is not a firmware dump, but OpenBeken release firmware. If you didn't capture a proper dump and have already flashed OpenBeken, it's not possible to capture anymore. In this case, please close the issue.

Drewix, 2023-09-05:
Correct dump

[ltchiptool_bk72xx_2023-09-05_14-49-39.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12523738/ltchiptool_bk72xx_2023-09-05_14-49-39.zip)

Cossid, 2023-09-05:
Added as `MOES MS-103 Temperature and Humidity Smart Switch`

closed 2023-09-05 by Cossid

divadiow, 2024-10-25:
OpenBeken now supports the DS18B20 sensor

kuba2k2, 2024-10-25:
I believe LibreTiny also should support it. DS18B20 is 1Wire, it's not something that has a hardware peripheral, so if ESPHome has a component for it, it should work.