tuya-cloudcutter/tuya-cloudcutter #459: New device - Aubess Power Monitor Switch

rafkan20, 2023-08-24:
Link to the product:
[https://www.aubess.net/products/aubess-wifi-mini-smart-switch16a-with-power-monitor-supporte-amazon-alexa-google-home-yandex-alice](url)

Dump file made with Beken writer 1.60:
[bk7231s_dump-2023- 8-24-16-34- 4.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12430649/bk7231s_dump-2023-.8-24-16-34-.4.zip)

rafkan20, 2023-09-02:
I still have a screenshot of the program BK7231Flasher
[readResult_BK7231N_QIO_Aubess_powe_monitor_2023-24-8-17-06-10.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12504487/readResult_BK7231N_QIO_Aubess_powe_monitor_2023-24-8-17-06-10.zip)

Cossid, 2023-09-05:
This firmware is patched and not exploitable via Tuya-CloudCutter.  Serial flashing will remain the only option for this device.

closed 2023-09-05 by Cossid