tuya-cloudcutter/tuya-cloudcutter #460: Cannot exploit Aubess 16A Mini Smart Switch

Slonik, 2023-08-25:
Hi!
```
root@ubuntu-test:/var/tuya/tuya-cloudcutter# ./tuya-cloudcutter.sh -s RTL579 xyz56
[!] WARNING: Selected wifi AP support: да
AP support is manditory for tuya-cloudcutter to work.  If this is blank or 'no' your adapter doesn't support this feature.
Press any key to continue, or CTRL+C to exitBuilding cloudcutter docker image
[+] Building 2.0s (13/13) FINISHED                                                                                      docker:default
 => [internal] load build definition from Dockerfile                                                                              0.0s
 => => transferring dockerfile: 460B                                                                                              0.0s
 => [internal] load .dockerignore                                                                                                 0.0s
 => => transferring context: 2B                                                                                                   0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-buster                                                      1.8s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:830e161433edfe047a23ebc99c12ee0eb1dc0a50e6b5f1c98e869ac27  0.0s
 => [internal] load build context                                                                                                 0.0s
 => => transferring context: 3.02kB                                                                                               0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-ess  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                     0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                               0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                          0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                      0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                        0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                         0.0s
 => exporting to image                                                                                                            0.0s
 => => exporting layers                                                                                                           0.0s
 => => writing image sha256:2c53e1168c5541527dd3cc8dcf2232c53108f0b5a2c817490dada460cfea593c                                      0.0s
 => => naming to docker.io/library/cloudcutter                                                                                    0.0s
Successfully built docker image
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Aubess
   Antela
   Aoycocr
   Arlec
   Atarm
   Athom
   Atomi
 > Aubess
   Avatar
   Avatto
   BNETA IoT
   Baytion
   Beca
   Biltema

[?] Select the article number of your device: 16A Mini Smart Switch
 > 16A Mini Smart Switch
   20A Plug
   IR Remote Controller

[?] Select the firmware version and name: 1.2.1 - BK7231N / oem_bk7231n_control_switch
 > 1.2.1 - BK7231N / oem_bk7231n_control_switch

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 5391.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: aubess-16a-mini-smart-switch
Selected Profile: oem-bk7231n-control-switch-1.2.1-sdk-2.3.3-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
......
Found access point name: "SmartLife-7E41", trying to connect...
Устройство «wlp2s0» успешно активировано с «7e1c276b-215e-442b-bc4b-d2549b5d318f».
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/Csj85TqAk8EL.deviceconfig
Saved device config in /work/configured-devices/Csj85TqAk8EL.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP

Found access point name: "SmartLife-7E41", trying to connect...
Устройство «wlp2s0» успешно активировано с «7e1c276b-215e-442b-bc4b-d2549b5d318f».
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```
What am I doing wrong?

ferbulous, 2023-08-28:
``` > 1.2.1 - BK7231N ```
Is this the same firmware version for the Aubess Switch?

kuba2k2, 2023-08-28:
You need to check the firmware version of the switch. There are a few guides and videos on how to complete the cloudcutter process.

ummetcivi, 2023-09-02:
I have the same problem. When I checked the FW version, it's v1.3.10. It seems it's not possible at the moment.

Cossid, 2023-09-05:
Yep, there are multiple physical and firmware versions.  You must use the one that matches your device.  If it is 1.3.10, the device is patched and can only be serially flashed.

closed 2023-09-05 by Cossid

pafade89, 2024-01-15:
@Cossid 

The fact that version 1.3.10 is not flashable is a general issue, or is there still a possibility that this could change?

Cheers P

Cossid, 2024-01-15:
1.3.10 cannot be exploited by CloudCutter, as the vulnerability is patched.  You can still serial flash those devices.

pafade89, 2024-01-15:
@Cossid 

Thank you very much for your super fast response. In that case, I will flash it serially.