tuya-cloudcutter/tuya-cloudcutter #463: Device does not connect to cloudcutterflash network

Gandalf1783, 2023-08-28:
My Tuya-Device (a so called "BRILO WIFI+BLE GU10") is (currently) not available through any profile.

I tried flashing it using the search via version feature.
The current version is 1.3.19, which is according to the tool a "BK7231N / oem_bk7231n_light_ty" profile.

This could be the case, it is a Wifi/Bluetooth light with CT control and RGB.

I have the complete terminal output of my (last) attempt here: https://pastebin.com/WeSSy5e9

As you can see, the device can perfectly connect in the attempts to exploit (e.g. "Exploit run, saved config file too!", "Flashing custom firmware..." lets me believe it works).
However, the last step is not moving forwards. 

The light does not connect and is constantly shining white light without blinking. 
Also, the script seems to wait for some time (but not more than 100 secs) after the line "Using WLAN adapater: wlp0s20f3" at the bottom of the log. It waits some time and is then printing the lines containing "Config file: /dev/stdin" etc.

The device I use to run cloudcutter is a Dell XPS17, running latest Arch Linux. 
I replaced the "service NetworkManager restart" with "systemctl restart NetworkManager".

 AFAIK, this does not result in any errors (yet), though I can see a single error line in dmesg: "iwlwifi 0000:00:14.3: WRT: Invalid buffer destination". Journalctl -xe did not show anything related to the script.

I also tried to turn off and back on the light a single time after the log says that the Wifi cloudcutterflash has been established, in the hopes of it connecting, but failure. I waited for around 3-4 minutes without any information in the log before terminating the script.

I am certainly aware that I am using a "unsupported" operating system to run it, but I wanted to check if this could be related to the issue before installing another OS.

Gandalf1783, 2023-08-28:
BTW: In regards to the -v option, I could not find a log. In a previous attempt, I tried to use the -v option, but no log was generated. I assume that this would appear in the logs/ folder?

Gandalf1783, 2023-08-28:
As it turns out, I did not wait long enough for the -v option to take effect.
I have a new log here: https://pastebin.com/tAS0Wg98

Gandalf1783, 2023-08-28:
I have opened one device carefully and was able to confirm that it is indeed a BK7231N. It has another printed silkscreen on it "CU110LFD". I will try to dump the firmware if possible tomorrow.
Please note this one of the devices where I **was not** able to flash it with ESPHome or get any further than what above. It seems to have changed the AP-Name. Also, I  cannot find it in the Tuya Smart App anymore. However, it still behaves like the original firmware. (e.g. 3x EZ-Mode, 6x AP-Mode for pairing)

kuba2k2, 2023-08-28:
Your logs show that you chose "detach" mode only, not flash. You need to choose option #2 to flash ESPHome to the device.

The AP name changed, which means the profile you used is correct.

Gandalf1783, 2023-08-28:
Correct, I chose detach to further test if it got past that point where it reconnects to the cloudcutterflash network. It does, like shown in the last log, but it does only seem to sit there.

AFAIK, you said on other issues that this should only take a few minutes, which does not seem to work out.

Can I try anything else to get these flashed or at least available via my IoT network?
I can also flash them if I'm not wrong, I got access to the Serial-Interface (RX1 and TX1, aswell as additional TX2), if nothing works out. 

I would like to avoid it, mainly due to the fact that I have to do this 6 times and on each bulb the PCBs or outer shell could crack (plastics, yaaay).

mkjustuk, 2023-08-29:
I've been trying to flash a few Aubess 16a switch boxes and I'm having the same issue, regardless of choose ESPHome or OpenBeken, it gets to the below point the same and then stops forever.  The AP has changed, so the exploit was done successfully with the options chosen, but no firmware flashing takes place.

wlp0s20f3: AP-ENABLED

Gandalf1783, 2023-08-29:
I assume that only devices that actually send a update request incase of CFW can actually get flashed. Maybe they have some kind of cycle? (every 24 hours a check runs?)

I am clueless btw on the simple "cutting" method. It seems like something else is missing...

Cossid, 2023-09-05:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

Gandalf1783, 2023-09-06:
I got the bulb right next to me and my notebook, they're not even a meter apart 

I would have tried this, I also read somewhere in a different issue that manually connecting the bulb through the app solved it.

However, I tried that, and neither in AP Mode, nor in EZ Mode I could get it to work. The Bulb no longer appears in the discovery function. My other lights, which I didn't temper with are still configurable via the tuya smart app.

Cossid, 2023-09-06:
You're only doing the process of add, the text also explicitly says it will not add to smart life.  If your device already has an `A-` AP prefix, it is impossible to add it to Smart Life, but you can still use Smart Life to trigger attempting, it just won't join smart life, but will work with other things listening like CloudCutter.

Yes, Smart Life will just scan endlessly, but it's sending information to the device that tells it to join the wifi network, which should be set in Smart Life matching the instructions in the FAQ.  I don't know exactly what it looks like in fast-blink mode, I still recommend slow-blink mode for this process, but fast-blink should technically work at this stage.

Gandalf1783, 2023-09-10:
So, do I understand correctly?

The tuya app will not see the device, but the mere search for devices in pairing mode will lead the bulb to reconnect to the cloudcutterflash network?

"the text also explicitly says it will not add to smart life"
-> I understand, it wont add it to the connected devices or let you control the bulb. I define "add" as in "adding the entry of the device in the smartlife app so it can be controlled (over the cloud)". Do we just have a bad misunderstanding? That (of course) cannot happen, I mean we literally want to disconnect them from the cloud, therefore I understand that any attempt to configure these bulbs should or will fail.

"Yes, Smart Life will just scan endlessly, but it's sending information to the device that tells it to join the wifi network,"
-> So even when scanning, even if the app does not "see" the bulb as in "I cannot find the device in the smartlife-app when scanning for devices in pairing mode", it still sends the command to connect to the last wifi / in this case the cloudcutterflash network?

=== IT WORKED ===

Before posting this, I tried it again.

I was not able to use the EZ-Mode to connect the bulb to cloudcutterflash.

I still have to replicate this behaviour - I could not accurately tell what I actually did because the SmartLife App is weird...

What my ultimate path was (so I suppose), was to click Add (the plus sign) in the top right of the iOS app, then manually select the lighting section and select the "generic" light option with BLE+WIFI (In german it was called "Lichtquelle (BLE+WIFI)".

Then I clicked on "Make sure the light is flashing or pulsing" and I chose "slow pulse/flash".
After that, I connected to the cloudcutterflash network on my phone, entered the cloudcutterflash password in the UI and clicked next, connected my phone to the A-XXX SSID that the bulb exposes and clicked next

After this, it should try to search the bulb (a big search-icon in the middle of the screen) and connect to the cloudcutterflash network. 

I cannot say why, but originally it did connect to the ccf-AP from my notebook but did not perform the update/connection to the cloud, when done through the app it seemed to work in just a few seconds.

TLDR; Try selecting the device manually in the SmartLife app, and do a "recovery" there to the ccf network.

I will not close the issue yet, since I have more bulbs and want to document the exact procedure in this issue for others aswell <3

Gandalf1783, 2023-09-11:
I was able to confirm that the above is the case.

I used AP Mode, since in EZ Mode nothing showed up.

If theres an further questions feel free to contact me

closed 2023-09-11 by Gandalf1783