tuya-cloudcutter/tuya-cloudcutter #465: Stalls after setting up the AP at the firmware flash stage.

mkjustuk, 2023-08-30:
Trying to flash a 1.2.1 switch module (one of several variants that all have the same issue).
I get past the AP stage (see below) no problem, but at the point of flashing firmware it seems to get stuck in a 5 minute loop - anyone see why?

`justin@192.168.1.122's password: 
Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Aug 30 18:03:40 2023
justin@raspberrypi:~ $ cd tuya-cloudcutter
justin@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -v
Building cloudcutter docker image
[+] Building 63.5s (13/13) FINISHED                              docker:default
 => [internal] load build definition from Dockerfile                       1.3s
 => => transferring dockerfile: 460B                                       0.0s
 => [internal] load .dockerignore                                          1.3s
 => => transferring context: 2B                                            0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-bu  28.6s
 => [internal] load build context                                          2.6s
 => => transferring context: 3.02kB                                        0.1s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:83  0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv              0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                        0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                   0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipe  0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                 0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                  0.0s
 => exporting to image                                                     0.1s
 => => exporting layers                                                    0.0s
 => => writing image sha256:a8bb7c288ee5c802f7a8121de052145d1a073d900279a  0.0s
 => => naming to docker.io/library/cloudcutter                             0.1s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 0.0.2 - BK7231N / oem_bk7231n_light_st[?] Select the firmware version and name: 0.0.3 - BK7231N / oem_bk7231n_light_st[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_ceiling_[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_dltj_plu[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_garage_d[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_jiabitai[?] Select the firmware version and name: 1.0.0 - BK7231T / HYS_03_018_bk7231t_t[?] Select the firmware version and name: 1.0.0 - BK7231T / oem_bk7231s_dltj_1pl[?] Select the firmware version and name: 1.0.1 - BK7231N / bk_light_rgbcw_vpylf[?] Select the firmware version and name: 1.0.1 - BK7231N / oem_bk7231n_scene_sw[?] Select the firmware version and name: 1.0.1 - BK7231N / oem_bk7231n_strip_ff[?] Select the firmware version and name: 1.0.1 - BK7231T / HYS_03_010_RGB_soq7i[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_1_plug_a[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_light_mi[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_ty_ffc_n[?] Select the firmware version and name: 1.0.10 - BK7231N / bk7231n_lowpower_co[?] Select the firmware version and name: 1.0.10 - BK7231N / oem_bk7231n_gate_ma[?] Select the firmware version and name: 1.0.10 - BK7231N / oem_bk7231n_light_p[?] Select the firmware version and name: 1.0.10 - BK7231T / oem_bk7231s_light_t[?] Select the firmware version and name: 1.0.14 - BK7231N / oem_bk7231n_slide_s[?] Select the firmware version and name: 1.0.15 - BK7231N / oem_bk7231n_slide_s[?] Select the firmware version and name: 1.0.18 - BK7231N / oem_bk7231n_slide_s[?] Select the firmware version and name: 1.0.2 - BK7231N / bk7231n_common_user_[?] Select the firmware version and name: 1.0.2 - BK7231N / oem_bk7231n_dctrl_sw[?] Select the firmware version and name: 1.0.2 - BK7231N / oem_bk7231n_strip_sl[?] Select the firmware version and name: 1.0.2 - BK7231T / HYS_03_010_RGB_xhsx0[?] Select the firmware version and name: 1.0.2 - BK7231T / bk7231t_common_user_[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_2_plug_g[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_control_[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_dltj_tes[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_light_mi[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_rnd_zero[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_strip_ke[?] Select the firmware version and name: 1.0.3 - BK7231N / oem_bk7231n_curtain_[?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_control_[?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_light_ty[?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_light4_4[?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_ty_ffc_n[?] Select the firmware version and name: 1.0.5 - BK7231N / oem_bk7231n_dctrl_sw[?] Select the firmware version and name: 1.0.5 - BK7231N / oem_bk7231n_light_at[?] Select the firmware version and name: 1.0.6 - BK7231N / oem_bk7231n_strip_rg[?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_dltj_2sw[?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_light_db[?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.0.7 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.0.7 - BK7231T / oem_bk7231s_rnd_zero[?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_light2_p[?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_light_db[?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_ty_ffc_d[?] Select the firmware version and name: 1.0.9 - BK7231N / BL_RGBCW_LIGHT_12KEY[?] Select the firmware version and name: 1.0.9 - BK7231T / oem_bk7231s_ty_ffc_d[?] Select the firmware version and name: 1.1.0 - BK7231N / oem_bk7231n_curtain_[?] Select the firmware version and name: 1.1.0 - BK7231N / oem_bk7231n_dctrl_sw[?] Select the firmware version and name: 1.1.0 - BK7231T / HYS_03_018_bk7231t_q[?] Select the firmware version and name: 1.1.0 - BK7231T / HYS_03_018_bk7231t_t[?] Select the firmware version and name: 1.1.0 - BK7231T / oem_bk7231s_curtain_[?] Select the firmware version and name: 1.1.0 - BK7231T / oem_bk7231s_dltj_out[?] Select the firmware version and name: 1.1.0 - BK7231T / oem_bk7231s_light5_s[?] Select the firmware version and name: 1.1.0 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.1.1 - BK7231N / oem_bk7231n_ffc_remo[?] Select the firmware version and name: 1.1.1 - BK7231T / HYS_03_018_bk7231t_a[?] Select the firmware version and name: 1.1.1 - BK7231T / oem_bk7231s_light_ty[?] Select the firmware version and name: 1.1.11 - BK7231N / oem_bk7231n_temp_hu[?] Select the firmware version and name: 1.1.14 - BK7231T / oem_bk7231s_switch1[?] Select the firmware version and name: 1.1.17 - BK7231N / oem_bk7231n_temp_hu[?] Select the firmware version and name: 1.1.2 - BK7231N / oem_bk7231n_control_[?] Select the firmware version and name: 1.1.2 - BK7231N / oem_bk7231n_dimmer_k[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231s_control_[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231s_hailai_1[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware ver [... cut ...]
Aug 30 16:29:24 dnsmasq-dhcp[15]: 1941713075 sent size:  4 option: 58 T1  6h
Aug 30 16:29:24 dnsmasq-dhcp[15]: 1941713075 sent size:  4 option: 59 T2  10h30m
Aug 30 16:29:24 dnsmasq-dhcp[15]: 1941713075 sent size:  4 option:  1 netmask  255.255.255.0
Aug 30 16:29:24 dnsmasq-dhcp[15]: 1941713075 sent size:  4 option: 28 broadcast  10.42.42.255
Aug 30 16:29:24 dnsmasq-dhcp[15]: 1941713075 sent size:  4 option:  3 router  10.42.42.1
Aug 30 16:29:24 dnsmasq-dhcp[15]: 1941713075 sent size:  4 option:  6 dns-server  10.42.42.1

`

Cossid, 2023-09-05:
Make sure your CloudCutter and target device are as near to each other as possible and follow this guide: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

Cossid, 2023-09-12:
Closing as stale.

closed 2023-09-12 by Cossid