tuya-cloudcutter/tuya-cloudcutter #466: New Device - Outon S1 Floor Lamp

brettmiller, 2023-09-04:
Outon S1 RGBCW floor lamp and RGB strip

Firmware Dump:
[Outon_S1-MF18228-RGBCW-floor-lamp-and-RGB-strip.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12511705/Outon_S1-MF18228-RGBCW-floor-lamp-and-RGB-strip.bin.zip)

closed 2023-09-06 by Cossid

divadiow, 2024-11-24:
v1.1.3 posted offers to update to 1.1.5
[1.1.5_boot.txt](https://github.com/user-attachments/files/17894678/1.1.5_boot.txt)
[1.1.5_Outon_S1-MF18228-RGBCW-floor-lamp-and-RGB-strip.zip](https://github.com/user-attachments/files/17894682/1.1.5_Outon_S1-MF18228-RGBCW-floor-lamp-and-RGB-strip.zip)
[image]

Cossid, 2024-11-24:
Added version to old, added profile for new version.