tuya-cloudcutter/tuya-cloudcutter #476: lumary-us-sd4e YAML Creation Issue

RetroVape, 2023-09-09:
I have lumary-us-sd4e but the 6in version. I was able to flash successfully, however not getting all the option there should be. 
Says "Module 'cmod': rgbcw doesn't match platform cwww"

[image]

Not sure where to go from here.

Cossid, 2023-09-09:
This is a bit beyond the scope of CloudCutter, as you're now dealing with 3rd party firmware.  This device is "RGB AI", which is another name for RGBIC, which I'm not sure is supported by LibreTiny/ESPHome yet, as it needs support added to a 3rd party component fastled.

closed 2023-09-09 by Cossid

RetroVape, 2023-09-10:
> This is a bit beyond the scope of CloudCutter, as you're now dealing with 3rd party firmware. This device is "RGB AI", which is another name for RGBIC, which I'm not sure is supported by LibreTiny/ESPHome yet, as it needs support added to a 3rd party component ~fastled~ neopixel.
> 
> You will need the data in Device configuration to configure the rgbic once it is supported.

This device has a profile on this repository, so pretty misleading if it’s not supported. That profile should be removed…

kuba2k2, 2023-09-10:
It is supported by Cloudcutter - you were even able to flash it successfully.
It is also supported for local control ("detach" mode instead of flashing).
Keep in mind that Cloudcutter and UPK2ESPHome or LibreTiny are completely different things. UPK is just a program that generates suggested config files, *based on* Cloudcutter profiles - they're not tied in any way.

Cossid, 2023-09-10:
> This device has a profile on this repository, so pretty misleading if it’s not supported. That profile should be removed…

CloudCutter supports either cutting or flashing of the device, but it does not support any 3rd party firmware you decide to install, or the configuration afterwards.  You should use the support vehicles of those respective firmwares instead.  This issue section is just for issues related to the cutting process itself.  We can't test the functionality of all features of all firmwares.  This is why the cutting option exists for where 3rd party firmware doesn't have full functionality.  Unfortunately, there is no going back once you've flashed 3rd party firmware however.

RetroVape, 2023-09-10:
> CloudCutter supports either cutting or flashing of the device, but it does not support any 3rd party firmware you decide to install, or the configuration afterwards. You should use the support [...]

That makes sense